wled/WLED #1123: Unable to compile, library id 359

pczekalski, 2020-08-30:
**Describe the bug**
I'm trying to compile for esp32dev, getting the message:
`LibraryManager: Installing id=359 @ 697c75a025
UndefinedPackageVersion: Could not find a version that satisfies the requirement '697c75a025' for your system 'linux_x86_64':`

**To Reproduce**
Import sources folder into the Visual Studio Code

**Expected behavior**
Compiles for esp32dev

**WLED version**
 - Board: ESP#@ DEVKITC v4
 - Version  0.10.0
 - Format self compiled

**Additional context**
The OS is Ubuntu 20.04 LTS x86_64, Platformio Code 4.3.4, Home 3.2.3, VSCode 1.48.2

Thank you for your help!

Aircoookie, 2020-08-30:
Hi!
For some reason the version of this specific commit for the `EspAsyncUDP` library is not found in some environments.
This has been fixed in master, I recommend compiling that.
If you want to compile v0.10.0, the fix is simple, remove `@697c75a025` after `ESPAsyncUDP` in your platformio.ini :)

pczekalski, 2020-09-02:
Thanks for the reply. I didn't make it to compile so far (removing version enforcer now causes my Platformio unable to find the library at all) but binary package works OK.

closed 2021-01-08 by stale[bot]