wled/WLED #1124: Restart when wifi is not available for longer period

feraldruid22, 2020-08-30:
This is not an issue with the project itself but with the ESP8266 D1 mini. 
I use these boards for various projects not just for WLED and they are always proven to be unreliable on the long run.

I had this board running WLED 0.10 since 28 days, in that time period my wifi router was also restarted once, the board reconnected no problem, however one day it just disconnected for good. The board itself still runs the effect fine but it is no longer reachable anymore on wifi.

I would suggest a wifi connection check feature which is if it detects that the wifi connection is down for a certain time it tries to reconnect if it cant it reboot the whole board with ESP.reset. 

I will now try it with disabling the wifi power saving but doubt it will help much. Sooner or later the board probably going to crash again. Since this hw is quite complex it is might be a partial kernel crash related to the networking code that is why the best workaround for now to just reset it.

Aircoookie, 2020-09-02:
Hi, yes this seems to be an issue with ESPs in general. I've experienced it several times in the past as well but lately it has been better for some reason (might be router dependent as well)

> I would suggest a wifi connection check feature which is if it detects that the wifi connection is down for a certain time it tries to reconnect

We are actually doing this. It doesn't seem effective though as even if the board is no longer reachable it appears to still be connected (it is in the router overview and the WLED UDP sync might work, but neither a simple ping nor port 80 will give a response). I've got yet to figure something out :) (now I remember though that there is [this usermod](https://github.com/Aircoookie/WLED/tree/master/usermods/Fix_unreachable_netservices_v2) which is designed to alleviate the issue. Might help!)

feraldruid22, 2020-09-19:
Hello,

Yes there are actually bunch of issues with the underlying ESP8266WiFi library but the team is still working on it and it is improving slowly. What I do usually in my ESP8266 sketches is something like this:

```
  while (WiFi.status() != WL_CONNECTED) {
    delay(200);
    Serial.print(".");

  conn++;

  if(conn==30)
    ESP.restart();

}
```

The issue you referred to about it thinks it's connected is because of the static ip. The previous check will always think it is connected IF static IP is used however it works as expected IF dhcp is used. I hope they gonna fix this up. You could implement a check like ping the gateways IP or if NTP is enabled then regularly query the NTP server (including at start) to see if you are connected or not.

As mentioned since there are bunch of other issues like memory corruptions on the long term it is better to completely restart the device if it can't reconnect to the wifi for a certain time period but sadly not even this is 100% foolproof because I have like 20 of these Wemos devices running and some even does go into a crazy restart loop for some reason and it just keep resetting until disconnected from the power and reconnected :(

ulte, 2020-10-22:
> Hi, yes this seems to be an issue with ESPs in general. I've experienced it several times in the past as well but lately it has been better for some reason (might be router dependent as well)
>
>
> We [...]

@Aircoookie I have the wled dropping wifi issue as well. I'd like to use that usermod you referred to yet not sure how to implement it. Do I make those file changes to wled00 directory and usermods_list.cpp file prior to flashing it?