wled/WLED #1128: Sunrise effect MQTT trigger

hell1337hound, 2020-09-01:
Hey Aricoookie, bloody amazing work mate! My sunrise MCU was causing issues and you've put sunrise into your latest build which is amazing. I also really appreciate how the red stage slowly spreads across the led strip!
I'd like to trigger it to wake me up to which i'm thinking i could do with an API call over MQTT but how would i get the effect to only run once? i notice once the sun has "risen" it runs the effect in reverse
Only other thing i can say is it would be great to have a second effect which is a super slow sunrise that maybe runs 3x slower as i know the speed slider can only affect things so far. So that folks have a choice between standard sunrise and super slow sunrise
Thanks!!

Aircoookie, 2020-09-01:
Hi, awesome that you like the new sunrise!
For your usecase, running the effect directly is probably not what you want.
Rather, setup the nightlight in LED settings to the `Sunrise` mode, whatever duration you want and target brightness 255.
Then, you can send the AP call `NL=1&ND` via MQTT or save it as a macro to activate at a set time in Time & Macro settings!
It will slowly ramp up and after the time has elapsed, will stay fully illuminated.
Hope it helps and have fun with it :)

hell1337hound, 2020-09-02:
will test once i get MQTT working and then close if successful

hell1337hound, 2020-09-03:
got this working but found the only way was to trigger &M=1 rather than send NL=1&ND

rudyo85, 2020-09-07:
hi hell1337hound, I’m trying to achieve the same. Can you tell me what you’ve configured where?

hell1337hound, 2020-09-08:
Hey Rudyo85, Go to your controller via app or web then Config > LED preferences
Set the "timed Light" to sunrise in the drop down and set the brightness to 255 (full brightness) and the "time" is how long you want the sun to rise
Then click save
Go to Time and Macros
in the 1# spot enter `NL=1&ND`

In your MQTT program use topic wledcontrollernamehere/api
payload `&M=1`

closed 2021-01-14 by stale[bot]