wled/WLED #1129: Unable to compile - 'class AsyncWebSocketClient' has no member named 'queueLength'

Alexxiioo, 2020-09-01:
While compiling a clean copy of WLED 10.2, i get this error when i try compiling on my Windows PC and also my MacBook. Am I doing something wrong?
Arduino: 1.8.13 (Mac OS X), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Legacy (new can return nullptr), All SSL ciphers (most compatible), 4MB (FS:2MB OTA:~1019KB), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

sketch/json.cpp: In function 'bool serveLiveLeds(AsyncWebServerRequest*, uint32_t)':
json.cpp:518:22: error: 'class AsyncWebSocketClient' has no member named 'queueLength'
     if (!wsc || wsc->queueLength() > 0) return false; //only send if queue free
                      ^
exit status 1
'class AsyncWebSocketClient' has no member named 'queueLength'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

sansillusion, 2020-09-01:
I have the same issue, I dont know what broke since I pulled latest esp32 Arduino, latest WLED and updated Arduino software.

Robotto, 2020-09-01:
ESPasyncWebserver has a canSend function that could probably do the job?

[https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.h#L229
](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncWebSocket.h#L229
)
So i'd replace line 518 in json.cpp with this:
`    if (!wsc || wsc->canSend()) return false; //only send if queue free`

Aircoookie, 2020-09-01:
Sorry for the confusion, I should have remembered to document this before releasing!

Websocket support (which is now enabled by default) requires [my fork of the EspAsyncWebserver library](https://github.com/Aircoookie/ESPAsyncWebServer)  It will also compile if you remove the `|| wsc->queueLength() > 0` part or if you disable websockets entirely in wled.h by commenting out `#define WLED_ENABLE_WEBSOCKETS`.

Added the info to the wiki and release!

blazoncek, 2022-07-25:
Closing as it may no longer be relevant. Please reopen if you think otherwise.

closed 2022-07-25 by blazoncek