wled/WLED #1130: Please add definitions to Wiki to reflect new attributes available with 0.10.2

DutchmanNL, 2020-09-01:
I would like to solve error's caused by new firmware for ioBroker integration which are triggered by new values in API : 

```
State attribute definition missing for + ws
State attribute definition missing for + mode
State attribute definition missing for + mi
```

I can't find a description for WS and mode, can you please let me now the purpose and what kind of data (Boolean/number) can be set here (also for mi please)

https://github.com/iobroker-community-adapters/ioBroker.wled/issues/38

DutchmanNL, 2020-09-01:
https://github.com/iobroker-community-adapters/ioBroker.wled/issues/38

Aircoookie, 2020-09-01:
Let you know in the linked ioBroker issue :)

closed 2020-09-01 by Aircoookie

DutchmanNL, 2020-09-01:
new version released : https://github.com/iobroker-community-adapters/ioBroker.wled/commit/e7e72ffdef6fea73a510bc921f4998d26611e19b