wled/WLED #1131: ledfx for groups

lechk82, 2020-09-01:
Is it possible to control a group with ledfx?
I have 4 strips with the length of 25 LEDS each, which I would like to control as a group with ledfx. But I can only control the whole 100 LEDs, so the effect is not the same as it would be for the group.

Thanks in advance.

tihoangyeudau, 2020-09-01:
How kind of your connection of them?

lechk82, 2020-09-01:
I have 100 Leds of type ws2812b in series connected to a pin of the wemos d1 mini (esp8266) which i divided in 4 segments within WLED with 25 LEDs each. the 4 segments work well together with wled effects, acting as a group simultaneously. I would like to create this behavior with ledfx, hence 100 LEDs with 4x25 simultaneously as a group. not the whole 100 as one strip.

datadrian, 2020-09-16:
make a device in ledfx with the specs for one of your strips, make the ip address have the last 3 numbers be 255.  this will control all ledfx on the network with one effect

Aircoookie, 2020-09-17:
Hi,
this is indeed not possible right now, streaming modes will only output to the entire strip, ignoring segments, which are only for built-in effects.
I will look into ways to make realtime modes (including LedFx/E1.31) to respect the segment configuration :)

franzuu, 2020-10-04:
Hi! Hoping to see this feature also. I also think it is important to respect the segment configuration and let E1.31 effects be displayed on a different part of the strip (maybe on segment 0?). I can already imagine all the possibilities 🤩 . Thank you for all the great work! 🙏

joshslater, 2021-03-17:
Would also love to see this feature!

raresserban, 2021-03-22:
Hey, interested in this as well :)
Not sure if maybe LedFX can also do something on their part, when you add devices to maybe specify from which leds you want to control them, so having segments in there.

StefvanEgmond, 2021-07-26:
Any updates on this yet? Would love to see it. Like @raresserban said, maybe it could be done by LedFX?

shauneccles, 2021-07-28:
LedFx can handle splitting segments/merging effects across multiple devices in our next release. If you jump onto LedFx discord and ask for a the beta we'll help you out - https://discord.gg/xyyHEquZKQ

blazoncek, 2022-10-10:
Streaming to main segment only is possible in 0.14

closed 2022-10-10 by blazoncek

im202055, 2022-12-23:
maybe try creating multiple e131 devices manually in ledfx with the same ip but different channel offsets? This seems to be working for me although its probably inefficient in some way.