wled/WLED #1132: Cant get V10.2 to light up

Rushmed, 2020-09-01:
Hello.

I was running successfull WLED 10.0 and previous version so several esp8266.
Now when updating to 10.2 just the first or three LEDs light up white. Tried USB and OTA to flash.
Webinterface is working and the setup is the same as on 10.0. 
Flashing back to 10.0 makes LEDs working again.
I have D4 as datatpin connected and flashed WLED_0.10.2_ESP8266_ledpin4.bin.
I also tried WLED_0.10.2_ESP8266_ledpin3.bin with D3 pin.

What am I doing wrong?

hell1337hound, 2020-09-02:
Hey @Rushmed have you tried setting it to less leds and seeing if more come on? like 10?
Are the 3 leds just stuck on in white and can't change?
are you using RGBW/RGB strip have you defs got that set correctly?

Aircoookie, 2020-09-02:
Hi, pin D4 is gpio2. You'll want to use the standard `WLED_0.10.2_ESP8266.bin` file :)

Rushmed, 2020-09-02:
> 
> 
> Hi, pin D4 is gpio2. You'll want to use the standard `WLED_0.10.2_ESP8266.bin` file :)

Thanks, that worked!

closed 2020-09-02 by Rushmed