wled/WLED #1135: esp-01 bin not retaining mqtt settings

hell1337hound, 2020-09-02:
Love your work <3, to the point:

MQTT settings dont retain at all.
flashed an esp8266 D1 mini with this bin:
WLED_0.10.2_ESP01_1M_ota.bin
Set wifi setting as i have before - gets an IP and retains those settings

Reason for choosing that bin was due to using the esp8266 small build with OTA which isnt available for 10.2

to reproduce issue flash that bin onto esp8266, click enable on MQTT and put settings in, hit save then go back to page

Have done this on the 9.x bin and it worked fine

Am getting WLED error 2 on the WLED_0.10.2_ESP8266.bin on an esp8266 d1 mini i suspect this is because my DI mini's are running out of memory hence using the small build.

Possible fixes: compile an esp8266 tiny build for the 10.2 distro 
check the esp-01 build on an esp-01 to see if it retains settings

Aircoookie, 2020-09-02:
Hi!
MQTT is disabled in `WLED_0.10.2_ESP01_1M_ota.bin`, otherwise it would use too much flash for OTA to work on 1mb chips. As a by-product, settings will also not retain as the parsing code for them is not included in this build. I plan on improving this soon so that `MQTT is not enabled in this build` is displayed instead of the non-functional settings fields.
You can use the `WLED_0.10.2_ESP01_1M_full.bin`, keep in mind though that it doesn't have OTA support.

If you are getting the WLED error 2 on a D1 mini with the default `WLED_0.10.2_ESP8266.bin`, your board is most likely not genuine/a ripoff where they have somehow only utilized an MB or two of flash instead of the 4mb that are standard for ESP-12F modules...

hell1337hound, 2020-09-02:
Ah ok, i'm guessing the changes in 10.2 stopped you from making an OTA build with MQTT because i was using the 10.0 minimal build with OTA with MQTT

hell1337hound, 2020-09-02:
I have the suspension i've been screwed! So this is what the esp8266 prints to serial after i've flashed it with the 1MB bin and THEN added my wifi settings. 
Would i be right in assuming that by writing the settings its gone over its memory and thus stopping it booting? 

`16:51:35.397 -> 
16:51:35.397 -> Exception (28):
16:51:35.397 -> epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
16:51:35.397 -> 
16:51:35.397 -> >>>stack>>>
16:51:35.397 -> 
16:51:35.397 -> ctx: cont
16:51:35.397 -> sp: 3ffffd20 end: 3fffffc0 offset: 0190
16:51:35.397 -> 3ffffeb0:  3fffff30 00000050 40100744 00001221  
16:51:35.397 -> 3ffffec0:  00000000 3fffdad0 3fff17f8 3fff1600  
16:51:35.397 -> 3ffffed0:  3fffff30 3fffff3c 00000000 4022fb5b  
16:51:35.444 -> 3ffffee0:  00000050 00000000 00000020 3fff13cb  
16:51:35.444 -> 3ffffef0:  00000003 3fff15ae 3fff0acc 3fffff3c  
16:51:35.444 -> 3fffff00:  3fffff30 00000050 3fff1600 4022fbc6  
16:51:35.444 -> 3fffff10:  3ffe9fcf 3fffff30 3ffe9fcb 3fff13cb  
16:51:35.444 -> 3fffff20:  3ffe9fcf 3ffe9fcb 3fff1600 4021e564  
16:51:35.444 -> 3fffff30:  00706374 6d01a8c0 83100744 64656c77  
16:51:35.444 -> 3fffff40:  3fff4900 84225dc4 40100700 3fff17f8  
16:51:35.444 -> 3fffff50:  3fff13d1 00000000 3fff07fc 4021e7e7  
16:51:35.444 -> 3fffff60:  00000000 62847c9d 00000000 3fff17f8  
16:51:35.444 -> 3fffff70:  00000000 00000000 00000000 3fff17f8  
16:51:35.444 -> 3fffff80:  3fffdad0 00000000 3fff07fc 4021e85f  
16:51:35.491 -> 3fffff90:  3fffdad0 00000000 3fff17b8 4021f23c  
16:51:35.491 -> 3fffffa0:  feefeffe feefeffe feefeffe 40236e54  
16:51:35.491 -> 3fffffb0:  feefeffe feefeffe 3ffe8824 401005b1  
16:51:35.491 -> <<<stack<<<
16:51:35.491 -> 
16:51:35.491 -> --------------- CUT HERE FOR EXCEPTION DECODER ---------------
16:51:35.491 -> 
16:51:35.491 ->  ets Jan  8 2013,rst cause:2, boot mode:(3,6)
16:51:35.491 -> 
16:51:35.538 -> load 0x4010f000, len 3584, room 16 
16:51:35.538 -> tail 0
16:51:35.538 -> chksum 0xb0
16:51:35.538 -> csum 0xb0
16:51:35.538 -> v2843a5ac
16:51:35.538 -> ~ld
16:51:35.584 -> Ada
`

hell1337hound, 2020-09-02:
Supposedly these should have 4mb of flash:
https://www.altronics.com.au/p/z6441-usb-mini-wifi-module/

Aircoookie, 2020-09-02:
The exception indicates that the board crashes once connecting to WiFi. This seems to happen sometimes with the `1m_full` binaries and I have yet to figure out what the exact cause is. Erasing the flash completely, then disabling Alexa emulation in Sync settings and mDNS in Wifi settings before entering the credentials is the workaround that works most of the time. Alexa is now disabled by default due to this issue, but since you were updating from 0.10.0 i assume it might have kept the old settings.

If you install esptool, you can use `esptool.py flash_id` to find out if you've been screwed. It will print out the actual detected flash size.

hell1337hound, 2020-09-02:
I've just tried another module with the 1mb bin and now that one is keeping the wifi settings but not connecting and staying in AP mode
So odd!

Aircoookie, 2020-09-03:
Hi, the issue that caused the exception after connecting for you yesterday should be fixed now! You can try whether the new binary `WLED_0.10.2_ESP01_1M_full_fix2009030.bin` from the release works for you :)

hell1337hound, 2020-09-03:
amazing thank you coookie! Will test soon

closed 2021-01-09 by stale[bot]