wled/WLED #1140: Automatic Philips Hue connect

Ronald-Baars, 2020-09-03:
**Is your feature request related to a problem? Please describe.**
No.

**Describe the solution you'd like**
-

**Describe alternatives you've considered**
-

**Additional context**
Every network that has a hue-bridge connected has an api available on https://discovery.meethue.com/ where it states the IP-address of the Hue-bridge. It could fetch this automatically to prevent the users from having to look for the IP themselves.

Thanks for making WLED. It's truly amazing!

huggy-d1, 2020-09-03:
Ok, so you are requesting adding a library supporting SSDP / UPnP protocol (used by Philips Hue Bridge) and logic + user interface changes required to support automatic discovery?

This is not a general linux package where this type of is available using a system service request.

I am not saying it is not possible because others have done it and shared their code on github.
I like the idea of having existing hue bulbs use WLED as the sync master (rather than the other way around).
It comes down to memory and available cpu time.  On an ESP32, this could work very smoothly because it can execute multi-threading (one core for regular WLED, and one core for HUE discovery), and has plenty of memory space available.

Ronald-Baars, 2020-10-15:
That sounds great, could be a real improvement!

closed 2020-10-15 by Ronald-Baars