wled/WLED #1141: No Disable WiFi sleep creates slow updates in Hyperion

sansillusion, 2020-09-04:
Hi, thanks again !

I have had no issues with my two esp32 with wled since I disabled current limiting in led setting. So I decided to remove the "Disable WiFi sleep" setting in wifi to see if I had crashes without it. I did not but I noticed that my Hyperion setup was very lagging, took me a while of debugging until I realized it might be that.

As soon that I check the checkbox and saved settings my Hyperion runs ultra smooth. Not really a bug but if someone else is having issues with very stiff/lagging wifi updates it's a good thing to try since it fixes all lags I get even with home assistant or plain web interface too.

Aircoookie, 2020-09-04:
Interesting, thank you for the insight!

My suspicion is that the sleep mode is too aggressive and will kick in way too early so that the module is in a continuous wake/sleep cycle. That is all good for occasional transmissions, but with realtime there is indeed a potential that wifi retransmissions become necessary or packets get dropped entirely which obviously causes lag.

sansillusion, 2020-09-04:
Yes sounds like exactly what's happening. If you want I can make a video to show the "huge" difference. But just so others know it's soooo smooth without sleep mode that it makes a world of difference on real-time reception.

kenlasko, 2021-06-03:
I know this is old, but I had been plagued with connectivity issues to my two Dig Quad deployments with ESP32 boards. For the most part, things would be OK, but I would see dropouts and laggy performance all the time.  Ping times would be all over the place (50ms then 700ms, then 300ms etc) and I would see periodic "No Reply" messages. I recently upgraded to the latest version of the Dig Uno with the external antenna thinking it was a wifi connectivity issue. When I still experienced the same issues, I dug deeper. As soon as I enabled "Disable WiFi Sleep", my issues went away. Ping times are now consistently under 5ms and everything is responding well.

sansillusion, 2021-06-03:
I personally think the default should be wifi sleep disabled. I am sure most wled setup rnu on power supplies anyway. and it would be easy to change for a battery setup.

Aircoookie, 2021-06-07:
I will set the new default to disabled on ESP32 and enabled on ESP8266.
On 8266, it quarters the power draw while there is a lot less decrease on ESP32. Considering these issues on 32, I believe the benefits of disabling it by default outweight the drawback of _maybe_ saving a tiny bit of power.

sansillusion, 2021-06-08:
I agree, and it still remains an option so if someone has the need for it, it can still be enabled.

blazoncek, 2024-06-23:
WiFi sleep will force AP to queue UDP packets (as discovered by @willmmiles) and as such will interfere with any UDP broadcast (i.e. Hyperion).

willmmiles, 2024-06-23:
We might consider adding an "auto" mode to wifi sleep that leaves it on normally, but disables it when any UDP sync mode is active.  It won't fix 100% of cases (since *every* node has to disable sleep to avoid having the AP buffer multicast packets) but it might cut back on the number of support cases.

I can sketch a PR when I get tired of fighting PWM issues ;)