wled/WLED #1142: Connect 3 stripes of WS2812B using wires - LEDs not working

thomas41587, 2020-09-04:
Hello,
I want to install the WS2812B LEDs on three sides of my bed. Between the stripes I need to have about 10cm of wire (because the LED stripe is to thick).
I do not think that you can do anything wrong here as it is just connecting three wires. But I rather guess there is some sort of technical limitation. The first stripe (attached to a wemos D1 mini) works perfectly. The second and third does not work. No LED turns on. 
I already verified that there is connection on all three wires and also the number of LEDs in WLED is configured properly. I also tried with additional 5V and GND cable directly from the powersupply but this did not change anything at all. 
[image]

Is there anything specific that needs to be added or changed if WS2812B need to be connected with wires?

Legsmaniac, 2020-09-04:
Is the direction to reach strip correct?

toto79, 2020-09-04:
Or the first led of the 2nd strip is defect. After a defect ws2812b no led will light up. (See the difference to ws2813/ws2815 with the data b line)
Your wiring seems to be ok, i see no special reason, why the trouble starts there. If it flickers, i would use a level shifter in front of the first led. I think it's an issue with the data cable, soldering or led.

nicr4wks, 2020-09-08:
Did you test the strip when it was in 1 piece?
If it worked as 1 long strip can you also try a very short piece of wire (1 inch max) between the strips.

thomas41587, 2020-09-08:
Thank you very much for your help.
I have tried several ways and all the ideas noted here. But it looks like the LEDs (or maybe the pcb strip itself) is very poor quality. I needed to cut out several LEDs and solder the ends together afterwards and ended up connecting the strip without wires. Instead I disassembled parts of my bed to get the strip between the wooden parts. Now everything seems to be working fine.
In the future I will check if there is any possibility to find better quality LED strips.

closed 2020-09-08 by thomas41587