wled/WLED #1143: esp32 ethernet

hell1337hound, 2020-09-04:
Hey Coookie I was wondering if it would be possible to put support for ethernet for the esp32 I know there's two different chips that are possible.

I've got one of these ethernet drivers and I could do some testing if you needed.

This would remove the need for WiFi in some situations that WiFi isn't feasible

Let me know what you think

thanks!

Aircoookie, 2020-09-04:
Yep, I believe this is a great idea as well for areas where WiFi is not feasible!
Already ordered a board with Ethernet for testing but haven't gotten around to implementing it yet. Hopefully soon :)

Closing as duplicate of #1027

closed 2020-09-04 by Aircoookie

vento17, 2023-05-11:
Hi Coookie
first of all, thanks for the greeeat tool you're sharing!
have you any news about artnet over ethernet?
would be wonderful!

blazoncek, 2023-05-12:
[KB](https://kno.wled.ge/interfaces/e1.31-dmx/)