wled/WLED #1144: Bug: Missing delay to activate sleep mode on ESP8266

c3n, 2020-09-04:
According to the ESP8266 docs you need delay(1) or more to enter DTIM-based wifi sleep (light sleep). Uploading the current version to a d1 mini without connecting anything else, the power consumption is very constant at 90mA, even after the 10s sleep timeout (turning LEDs off in web gui doesn't change that).

However if I change wled.cpp in line 89 to:
if(offMode)delay(1);else strip.service();

and then turn off the LEDs in the web gui the power consumption jumps between 35mA and 55mA (while the behaviour with LEDs on is unchanged)

I didn't make a pull request because I have no idea what timing limitations I broke with that delay statement

pbolduc, 2020-09-12:
> According to the ESP8266 docs 
It would be useful to link to the documentation so we dont have to go looking for it.

c3n, 2020-09-12:
https://blog.creations.de/?p=149 says that you need a delay of at least 1 to go into light sleep while the official docs https://www.espressif.com/sites/default/files/9b-esp8266-low_power_solutions_en_0.pdf just say that the cpu needs to be idle under 3.2.1.

pfeerick, 2020-09-14:
I also noticed that the custom D1 Mini setup I just converted over to WLED isn't sleeping - so is staying nice and warm. I can say from personal experience, and following the modem/light sleep issues that arrose the across esp8266-arduino core v2.5.x to  v2.7.x that auto light sleep requires a delay to operate, but it doesn't seem to be mentioned in the docs. It is however, specifically menmtioned in the low power example code provided with the core (my emphasis added):

> This is the default power saving mode when you have an active WiFi connection. You don't need to add anything to your code to get this mode. **_The only time the modem sleeps is when your program spends time in delay() frequently in STA mode._** Any delay() time works as long as it happens frequently. 

https://github.com/esp8266/Arduino/blob/master/libraries/esp8266/examples/LowPowerDemo/README.md#test-2---automatic-modem-sleep

pbolduc, 2020-09-14:
Do we know if this applies to ESP32 as well?

pfeerick, 2020-09-14:
I'm not familiar enough with the ESP32 to comment from experience. lastminuteengineers have an [excellent article on the ESP32 sleep modes](https://lastminuteengineers.com/esp32-sleep-modes-power-consumption/#esp32-modem-sleep), and whilst it appears the modem sleep works by sleeping between DTIM beacons like the ESP8266, there is no mention of a delay needed. 

But going by the ESP-IDF [example on wifi power saving](https://github.com/espressif/esp-idf/tree/master/examples/wifi/power_save) - it looks like it simply needs to be enabled - as it claims there that it is disabled by default. i.e. via `esp_wifi_set_ps(WIFI_PS_MIN_MODEM);`

But then, when looking at the SDK source code included with the esp32-arduino core, it then [says that WIFI_PS_MIN_MODEM is the default](https://github.com/espressif/arduino-esp32/blob/37a7fb3d6ada8a9230c96e2df999c95ef10fd4d3/tools/sdk/include/esp32/esp_wifi.h#L463)?! Now I really am confused!

pfeerick, 2020-09-14:
Hey @c3n, have you noticed that the ESP8266 won't stay in modem sleep mode? Mine seems to properly sleep for all of 5-7 seconds and then stays always on. Seems there is too much going on for the modem to keep napping. :(

I thought I'd be clever and integrate it into the 'no wifi sleep' option at the bottom of the wifi settings page, so was actively testing/toggling it on and off, and realised the failure to stay sleepy.

[image]

c3n, 2020-09-14:
Unfortunately I can't test anything for the next 2 weeks, but mine definitely slept a lot longer. That being said, the sleep mode seems a bit sketchy, it's supposed to wait for 10s (I think) before going to sleep which for me it usually (but not always) ignored and went straight to sleep after each web page action

Aircoookie, 2020-09-14:
When I was specifically testing modem sleep about a year ago I had my Wemos D1 idling at 20mA, with occasional spikes to 80mA for communication. On ESP32 sleep mode did not seem to work at all, with a constant 120mA of power draw. Looks like I will need to get the power meter and try to look into why the sleep seems to be no longer working...

pfeerick, 2020-09-14:
It's supposed to wait for 10 seconds after the initial connection to the
AP, due to loss of packets issues if it entered modem sleep immediately
after connection. So that sounds right.

I should probably revert to defaults, and start again, maybe turning
something on (i.e. Alexa notifications, etc) is keeping it busy. As ~30ma
on average was what I was expecting to see, rather than staying at ~140ma
all the time.

PR #1184 (2020-09-20): ESP8266: Allow auto modem sleep if power off and sleep permitted
Closes #1144. 

Adds a 1ms delay _only_ when main power is off and the `noWifiSleep` option is not active, and _only_ for the ESP8266. This is needed for the automatic modem sleep to operate on the ESP8266 between DTIM beacons. The ESP32 appears to have a different mechanism, via the `esp_wifi_set_ps()` API call, but that a problem for another issue/PR. 

I'm leaving this as a draft for now, until I've been able to test it for a few days for unforeseen side effects. ;)

To show the effect of this on my Wemos D1 Mini, I powered it [... cut]
closed 2020-09-20 by Aircoookie