wled/WLED #1146: JSON API over Serial

Aircoookie, 2020-09-04:
**Is your feature request related to a problem? Please describe.**
It would be nice to have easy wired control for WLED in addition to WiFi (and Ethernet once added)

**Describe the solution you'd like**
Add the capability to read JSON API commands from the Serial interface
(and optionally, to return the JSON state)

kantlivelong, 2021-08-16:
While a bit more complex it could be worth implementing SLIP ([RFC1055](https://datatracker.ietf.org/doc/html/rfc1055). This would allow using the existing WebUI/APIs over serial.

Aircoookie, 2021-08-16:
@kantlivelong interesting idea! The existing web UI could also be hosted on a web server and access the WLED device via the [Web Serial API](https://developer.mozilla.org/en-US/docs/Web/API/Web_Serial_API) :)

In fact this API is already in use on https://install.wled.me to flash WLED to an ESP!