wled/WLED #1147: Rain Animation is inverted

CSchlipp, 2020-09-05:
**Describe the bug**
The direction of effect "Rain" is inverted. Drops are moving up, instead of down.

**To Reproduce**
Use vertical LED strip, select effect "Rain". 
Do not reverse the LED order.

**Expected behavior**
Effect makes raindrops are falling from top to bottom.

**WLED version**
 - Board: Wemos D1
 - Version 0.10.2
 - Format Binary

**Additional context**
Reverse LED order (rotate 180) works, but then all other effects like "Fire 2012", "Fireworks 1D", ... are wrong.

jsloyer, 2020-11-04:
can you list the other effects that are affected?

CSchlipp, 2020-11-04:
As far as I can tell, Raindrop is the only reversed effect.
Fire 2012 or Fireworks 1D are just arbitrary examples of other effects that have a clear orientation and can thus be compared to the Raindrop effect orientation.

jsloyer, 2020-11-04:
ok, would you really want an inversed raindrop effect? I guess you could...

CSchlipp, 2020-11-04:
I'm not sure we're talking about the same things, so let me rephrase:
On a vertical LED strip, where animations like Fire2012 are showing correctly (Flames growing from bottom towards top), the Raindrop ankmation is wrong (Drops fallkng from Bottom to top, instead of from top to bottom).

Aircoookie, 2020-11-04:
Actually never ran a vertical strip, so didn't notice.
Changed the effect to run in the same direction as e.g. Drip and Fire 2012 now in latest source :)

commit b8fcab29 (2020-11-04): Inversed Rain direction (fixes #1147)
closed 2020-11-04 by Aircoookie