wled/WLED #1149: HTTP update

SpicerSolutions, 2020-09-05:
Is updating over HTTP planned (such as using ESP8266httpUpdate?). I appreciate I can probably add this with a usermod.

It'd be great to send out device updates to my team using a central server.

Thanks

Aircoookie, 2020-10-29:
Hi,
first of all I'm very sorry for the late response, got a bit overwhelmed recently :)

This would be a great feature to have as it would simplify the updating process for the user and would not require updating each device manually.
There are however a few security concerns, the last thing I want is someone uploading malware through an unsecured updater.
For ESP32 with solid crypto support, this is certainly possible and would allow for a comfortable and secure update. With ESP8266 I'm not sure at all though since the available heap memory is so low.
But yeah - a usermod is a good idea if it is used in an environment where the security implications are understood!

softhack007, 2023-01-06:
Closing, as we have OTA update. Also mass-updating via script or via HA is already possible.

closed 2023-01-06 by softhack007