wled/WLED #1150: Nightlight remaining time indicator

Aircoookie, 2020-09-05:
**Is your feature request related to a problem? Please describe.**
Time remaining on nightlight is unclear

**Describe the solution you'd like**
Add an indicator in the UI that displays the remaining time on the nightlight as a countdown

**Additional context**
Suggested by Virus_cz on Discord

catlee, 2020-10-31:
@Aircoookie do you have a preference how this is represented in the API? My PR adds a `rem` field to the `nl` status which represents the number of minutes remaining for the nightlight, or -1 when it's not active.

Aircoookie, 2020-11-04:
@catlee thank you for the PR, will merge as soon as you mark it ready. I like the `rem` property, would suggest though that we go with seconds for a bit more precision on the API side. Then it will be up to the client how to display the remaining time most intuitively.