wled/WLED #1155: Add usermod helper function for elapsed time with rollover handling

pbolduc, 2020-09-07:
I heard on DrZzs stream tonight that the Temperature usermod caused some flicker. I was making some changes to address that but wanted to handle a rollover safe `elapsed` time function.  I can see a number of the usermods do not use safe elapsed time calculations. After about 50 days, these usermods would stop reporting temperature due to rollover of the millis() function.  Since this seems to be a common operation, we should consider adding to the usermod base class or elsewhere.  Here is what I have come up with, totally not tested, but seems fairly logical.   FYI, my branch with the Temperature usermod changes are here:  https://github.com/  pbolduc/WLED/tree/async-temperature-usermod/usermods/Temperature expect a PR in the coming days after testing.

```c++
    /**
     * Calculates the elapsed milliseconds since the specified time.
     * Handles the case where millis() rolls over after roughly 49.7 days
     */
    unsigned long elapsed(unsigned long since) {
      unsigned long now = millis();

      // no rollover?
      if (since <= now) {
        return now - since;
      }

      // rollover in millis - happens after roughly 49.7 days
      return (UINT32_MAX - since) + now;
    }
```

Aircoookie, 2020-09-07:
Have you actually observed a problem with the rollover causing timers to stop working in some usermods?
This code will indeed stop working on rollover:  
`if (since + interval > now) action()`  
However,  
`if (now - since > interval) action()` is unaffected and will continue to work despite rollovers due to the nature of the subtraction also rolling over. This is what I use all throughout WLED and it never caused issues even beyond 50 days of uptime.  
https://oshlab.com/handle-millis-overflow-arduino/
  
Thank you for the usermod improvement, glad to see the library makers thought of a way to not have to block during the measurement. Feel free to open it as a PR against the existing Temperature usermod once you confirmed it works well :)

pbolduc, 2020-09-08:
Of course you are correct :) You are Aircoookie! I do most of my day job in C# where we typically dont use unsigned types.  I only speculated there would be an issue with rollover.  Now I can revert the period check.

The issue I am working on with Quindor, is to do with the Dallas Sensor and flickering when enabled.  The current code uses a blocking API and when the sensor is actually read, the code will block for 93-94 milliseconds (as per the datasheet with 9 bit resolution) to allow the sensor to complete the read. The library allows for an async non-blocking implementation. As the sensor is only read every 60 seconds, that cant be the cause of the flickering.   However, some of the changes I have made are in attempt to reduce the overhead.

closed 2020-09-08 by pbolduc