wled/WLED #1157: Server Descriptions Always Displayed

Araforn, 2020-09-07:
Hello.

I understand this might be specific to me but I have added a number of WLED servers that are receiving streams from hyperion.
Due to the nature of the setup, Hyperion always streams and therefore  i am consistently just seeing the streaming info on the app where the server description should be.

See image below.

[image]

I would like to see all my server names and perhaps an asterisk only to show that this server is receiving a stream. When i click in to that server I could then see the streaming source.

I know the recommendation is to stop the streaming service but I am happy with automations set up in home assistant that sends the JSON Api command to disable the stream.

Ignore if you feel this is totally unnecessary but I could modify the code for my own if you could point me in the right direction?.

Thanks a lot.

Best Regards,

Aircoookie, 2020-09-07:
Hi and welcome!
After a similar request it occured to me as well that the dynamic description causes this issue in the app if streams are running, so the behavior was changed recently :)
Make sure you update to WLED 0.10.2 and it will display the description as normal with just a ` (live)` appended if a stream is currently running.

Araforn, 2020-09-07:
Great. Thanks so much for the response. I'll update all my installations. Really great work with WLED.

closed 2020-09-27 by Aircoookie