wled/WLED #1160: Esp32s will not flash keep getting “ A fatal error has occurred:failed to connect to esp32: timed out waiting for pocket header”

superbond05, 2020-09-08:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

huggy-d1, 2020-09-08:
Are you able to write stock esp32 firmware to device?

It would help if you were to edit your post and transfer most of the title into the post, and then edit the default text with the details related to your specific situation.

huggy-d1, 2020-09-08:
This tutorial may help: [https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/](url)

superbond05, 2020-09-09:
I tried to rewrite to stock. Doesn’t matter what I do it always come up “fatal error waiting for pocket header” I’m sorry but I’m green as green can be when it comes to this kind of thing. I’m only here because I promised my wife new Xmas lights on the house. Don’t take this the wrong way. But I would like to learn how to do this. I just need to do baby steps at first. Hope you can help. If your too buy I understand

superbond05, 2020-09-09:
Thanks for trying but that didn’t work still get the same message 😩

On Sep 8, 2020, at 8:40 PM, Super <superbond05@hotmail.com> wrote:

 I tried to rewrite to stock. Doesn’t matter what I do it always come up “fatal error waiting for pocket header” I’m sorry but I’m green as green can be when it comes to this kind of thing. I’m only here because I promised my wife new Xmas lights on the house. Don’t take this the wrong way. But I would like to learn how to do this. I just need to do baby steps at first. Hope you can help. If your too buy I understand

superbond05, 2020-09-09:
Here is what I did and no change still error

Sent from my iPhone

huggy-d1, 2020-09-09:
Did you find a way to press Boot/Flash button at power up so you could enter flash mode like the link I posted?  Or, did you install a 10µF capacitor between EN and GND pins on the ESP32?

superbond05, 2020-09-09:
I soldered a wire from EN to the top left of the wifi chip as in the instructions. Tested it with a multimeter. Tried to flash it with the latest wled update 10.2. Still got that error message

Aircoookie, 2020-09-09:
Do you have a different ESP32 on hand? If keeping the boot button pressed during flashing does not work it might be a defective unit.

superbond05, 2020-09-09:
That’s my thinking also.
No. I should be getting some in the next day or two.
 Sounds like I’ll be needing five maybe more. Nice to know how to fix them just in case it happens again.
 Did you get the pics of what I did? To see if I did it correct

huggy-d1, 2020-09-09:
Your uploaded images are not getting displayed on github (at least not in Chrome/Windoze10).

superbond05, 2020-09-09:
![Uploading 92663678...]

superbond05, 2020-09-09:
[image]

superbond05, 2020-09-09:
[image]

superbond05, 2020-09-12:
So that one I gave up on. Bought 5 more and flashed them. Only one worked. With the help of wingscastona we figure it out. Thought I had no data coming from the esp32s but it turned out that I set the leds to 1000 in the wled configurations. Changed it to 450 and all is good

Aircoookie, 2020-09-27:
@superbond05 if the LEDs stay black when set to 1000 it is because of the current limiter. You  can uncheck `Enable automatic brightness limiter` in LED settings to get full brightness if your power supply is adequate :)

closed 2020-09-27 by Aircoookie