wled/WLED #1161: SYNc Problem

NakkeNakuttaja, 2020-09-08:
I have 3 WEMOS D1 mini controller, and i can't get those to sync. Do i need MQTT server that i can use sync feature. I have tried to every variats. Ports is same, receive UDP realtime is on, sync buttons is on, only reason what i came up is that i need MQTT server, any ideas anyone. WLED firmware is latest.

sansillusion, 2020-09-08:
I had a similar issue, turns out it was a router setting preventing my devices to talk together. In my router it was under wifi network interfaces settings, a setting named "isolate devices"

NakkeNakuttaja, 2020-09-08:
Thanks! That worked. I have check every settings everywhere, but that settings is under wireless advanced, where is transmit power and some other settings what you never need to change, and thats why i didn't even check that menu at all. Mine was Enable AP Isolation.

closed 2020-09-09 by Aircoookie