wled/WLED #1162: Sunrise 0.10.2

Yusef1985, 2020-09-08:
Hi Guys, 
Can anyone tell how I can set the sunrise effect to start in a preset time? 

Btw. @Aircoookie thank you very much for the new effect and in general for being awesome 👍

huggy-d1, 2020-09-08:
Is your NTP configuration working well?  Are you using a local NTP server, like your wifi router?  If so, you have a chance.
A simple solution would be to schedule a web request at a specific time of day (cron job or windows task scheduler).  
Use the web or json api to request WLED to configure itself for the sunrise effect, with starting brightness, and ending brightness, and a slow rate.
Or, schedule a new request every few minutes with a new brightness setting if WLED does not play along with what you want.
You can play around with the url / json to send so it gives you what you want in a web browser / json tester.

Many of us have some device that is always on, and can run a cron job (or schedule a task).  Many routers can execute cron jobs with a little effort.

Yusef1985, 2020-09-09:
@huggy-d1 
Oh man I thought it could be done via macros, I got it to work with macros but only once it could not repeat it it just started from where it stopped last. 
  
Do you have a guide on how to make this work?

Aircoookie, 2020-09-09:
While the external commands will work well (e.g. with a cron job like @huggy-d1 suggested), you can indeed make it work with timed macros!
Firstly, go to LED settings and in the Timed Light section, set the Nightlight mode to `Sunrise`, target brightness to 255 and your desired fade-up duration in minutes. Save and then go to Time & Macros settings.

Now, check the box `Get time from NTP server`, enter `NL=1&ND` in the first macro slot, and set that macro to in one of the timed light slots. The time you input is where the sunrise will start, not when it will reach full intensity. You can even use multiple timed light slots to set different times for some weekdays or make another macro that turns the LEDs off after some time :)

huggy-d1, 2020-09-09:
Thanks @Aircoookie for posting the better way!

Yusef1985, 2020-09-09:
Hi @Aircoookie Thank you very much for your reply. 
https://pasteboard.co/JqjEt4B.jpg
https://pasteboard.co/JqjEEOw.jpg

Can you tell me what am I doing wrong?

Aircoookie, 2020-09-09:
@Yusef1985 that looks exactly correct!
Two things that might cause it to not have worked in your test:
- Maybe the minute had already passed to 19:55 when the macros were saved
- Maybe the light was on before? The sunrise will only work as expected if starting with the light off, otherwise it will do a sunset actually

Yusef1985, 2020-09-09:
@Aircoookie it was off but the thing is it just starts in full brightness

Yusef1985, 2020-09-09:
do i have to reset it before turning it off?

Aircoookie, 2020-09-09:
No, just click the power button in the main UI, no need to cut the power. If it was suddenly turning on 19:56, probably the nightlight mode is still on "Wait and set" somehow

Yusef1985, 2020-09-09:
The issue I'm having is when the light turns off in surprise mode (full brightness ) next time it turns on it just turns on in full brightness without doing the sunrise animation

Yusef1985, 2020-09-09:
okay, I think I figured it out. I just have to set the light to a different effect before turning it off, then the sunrise animation starts correctly

huggy-d1, 2020-09-10:
What ntp server did you end up using that's both working and not causing wifi dropouts?

Aircoookie, 2020-09-11:
@huggy-d1 The default `0.wled.pool.ntp.org` works well for me despite being in Germany and a DNS lookup saying the corresponding IPs are all based in the US. `ntp.org` has a system where this DNS matches multiple IPs that also change to balance load. I believe the ESP would only ever use the first IP that is returmed by the DNS query, and if there happens to be an issue with that server, it will only work once the DNS record changes and the ESP is rebooted.
So it seems to be more of an YMMV thing than I hoped for, but if it doesn't work reliably for you, you could use the IP of a DNS server in your area.

@Yusef1985 in latest master, I fixed Sunrise not properly re-initializing if no other effect was set! For 0.10.2, setting FX=0 with the turn off macro is a very good workaround though!

closed 2020-09-27 by Aircoookie