wled/WLED #1168: WLED Json won't let me change anything inside state with python?

lukasdim, 2020-09-11:
**Describe the bug**
I'm not positive if this is a bug or if this is me, but every time I add state (ex. {"state": [{"bri": 255}]}) it says success but never really does anything I change inside the state.

**To Reproduce**
Using python requests, send a post to wled json client with state parameters. It will say success, but it ignored what you did.

**Expected behavior**
I expected python and/or the wled controller to take count of anything i changed inside of the state.

**WLED version**
0.10.2
 - Board: ESP8266
 - Version: 0.10.2
 - Format: Binary

**Additional context**
I am using python 3.8.3. Any help is appreciated.

Thank you for your help!

Aircoookie, 2020-09-11:
Hi!
Currently, every POST request you make to either the `/json` or `/json/state` endpoints is treated as the state object.
So you can just send `{"bri": 255}` and that should work!
I can consider also processing a request like `{"state":{"bri": 255}}`, because that will align with what is outputted by a GET request to `/json` (even though you can't change any value not in the state object) and it will also make the API just a little bit more REST-compliant.

lukasdim, 2020-09-11:
That worked, thank you!

closed 2020-09-11 by lukasdim