wled/WLED #1169: WLED does not save segments

athlon1900, 2020-09-11:
**WLED version**
 - Board  ESP-01S upgraded to 4Mb
 - Version 0.10.0 build 2005030
 - downloaded Binary

Problem:
Saving the segments is not possible.
Must always be set after booting.
Or is there hidden attitude ?

Thank you for your help!
athlon1900

Aircoookie, 2020-09-11:
Hi!
There is indeed a possibility to save segments permanently:
https://github.com/Aircoookie/WLED/wiki/FAQ#my-segments-are-gone-after-a-reboot

athlon1900, 2020-09-11:
Hi Aircoookie !
Thanks for the link and your answer.
I will try it.

athlon1900

chunkysteveo, 2020-11-24:
That just saved me going up a ladder and pulling out my ESP01! Thanks guys!!! I assumed it has some dodgy RAM chip on it, but this little "hack" worked. I did notice that note on slot 16 but didn't quiet understand what it was referring to.

@Aircoookie  -  it would be worth adding the FAQ answer you linked to for saving segments in the [Segment section](https://github.com/Aircoookie/WLED/wiki/Segments) as I (wrongly!) assumed they would save automatically and not lose them on boot.

Keep up the great work!

closed 2021-06-04 by stale[bot]

markormesher, 2022-12-31:
For anyone finding this since the GH wiki was deprecated recently, here is the new link: https://kno.wled.ge/basics/faq/#my-segments-are-gone-after-a-reboot