wled/WLED #1170: SPIFFs is deprecated

pille, 2020-09-11:
SPIFFs is not being worked on since 2017 and is now deprecated.
there's an almost drop in replacement called LittleFS.
the project should move to the new FS implementation, especially since the upcoming release plans to facilitate the flash more that before.

https://arduino-esp8266.readthedocs.io/en/latest/filesystem.html#spiffs-deprecation-warning

pbolduc, 2020-09-12:
Sorry, I dont agree with the label of bug. I would think this issue is more of an enhancement. The page you reference says SPIFFs is deprecated. This means we should move off of it, not that it is no implemented correctly.

pbolduc, 2020-09-12:
> They share a compatible API but have incompatible on-flash implementations, so it is important to choose one or the per project as attempting to mount a SPIFFS volume under LittleFS may result in a format operation and definitely will not preserve any files, and vice-versa.

This makes it difficult to migrate to without having a backup and restore function. If a new version of WLED moved to LittleFS, then the saved data on the flash file system would be lost. When people upgrade versions they dont expect their settings to be lost.

> The actual File and Dir objects returned from either filesystem behave in the same manner and documentation is applicable to both. To convert most applications from SPIFFS to LittleFS simply requires changing the SPIFFS.begin() to LittleFS.begin() and SPIFFS.open() to LittleFS.open() with the rest of the code remaining untouched.

This is promising as the code required to change should be minimal.

Aircoookie, 2020-09-12:
Thank you for bringing up this point. I agree that we should try to avoid SPIFFS since deprecation is not done likely and running un-maintained code with lots of documented, but unfixed issues could prove to be detrimental in the long run. `0.11.0` is the first WLED version that will utilize the Filesystem to a significant degree.

Currently I am working on storing Presets in FS and the code works flawlessly with both filesystems from the limited testing I've done so far.

The often mentioned benefit of LittleFS being faster is not something I can necessarily confirm - the operations used to add a new preset are 120ms on average running LittleFS instead of 25ms with SPIFFS but as long as any operation stays below 500ms we should be fine either way!

On ESP32 it is a different story since there is no LittleFS support built into the core yet ([but it could be coming soon!](https://github.com/espressif/arduino-esp32/pull/4096)), but if possible I want to use LittleFS by default on both platforms since I regard it as more future proof and I do not want to have to migrate to an incompatible filesystem down the line. Manual backup/restore via the filesystem `/edit` page is easy enough but a major inconvenience if necessary for updating multiple units.

pille, 2020-09-12:
just wanted to bring this topic up, _before_ you release FS support.
i already saw your future-proof `WLED_FS` definition :wink:

lorol, 2020-09-13:
You can try to plug https://github.com/lorol/LITTLEFS while waiting for blessed esp32 support within IDF and Arduino esp32 core. PlatformIO projects can also use it with little tweaks. 
https://github.com/rjwats/esp8266-react/issues/197

The functional difference with esp8266' LittleFS "folderless-SPIFFS-like" implementation is an eventual need to iterate and create/delete folders at application, on file creation and deletion if you want to use old code made for SPIFFS. But this can be done on later revisions if absolutely required.

pfeerick, 2020-09-20:
Until (and if, since their reason for not following the ESP8266 core is that the ESPIDF natively supports SPIFFS) the ESP32-Arduino gods depreciate SPIFFs, I don't see any real reason to use LittleFS over there - or can you make a data partition blob that can be flashed on both an ESP8266 *and* an ESP32? If that were the case, that alone would be a good reason to switch, especially not. Regardless, it is a good idea to use LittleFS on the ESP8266 now, since this is effectively new code and SPIFFS is depreciated, and AFAIK will be pulled in v3 of the ESP8266-Arduino core. Sticking with SPIFFS there will only result in later headaches, so better small ones now than bigger ones later!

lorol, 2020-09-20:
Too many useless words from your and my side.
Yes, l provided a mean to make and flash a data to LittleFS partition. The driver is through IDF and it is as much native as it could be at all. Just test you wish.

closed 2021-01-25 by stale[bot]