wled/WLED #1171: Random Cycle: Palette transitions

xxv, 2020-09-11:
Currently, when color palettes shift in the Random Cycle palette mode, it jumps hard from one palette to another. It'd be nicer if it smoothly transitioned between the two palettes.

For timing, Random Cycle already picks a time (from where?) so either using a fraction of that time or just a hard-coded value like 500ms would be fine. Alternatively, Random Cycle could be overhauled a bit more to include customizable timing somewhere in the settings.

This would likely involve creating a copy of the previous palette in memory, so there may be a (small?) memory impact from this feature. It would also involve recomputing the palette on every frame so if there are palette update costs, that could also be an issue.

(I'm happy to take charge of implementing this, but I wanted to create an issue to solicit suggestions/feedback before putting up a PR)

huggy-d1, 2020-09-11:
After reading your post, I have thoughts on the palette shift abruptness.

- Option 1: Leave as is.  Some people may expect / want this behavior
- Option 2: Hard-code transition time, that is visually pleasing to most everyone (the Apple way)
- Option 3: Auto-calculating transition time, based on Segment.speed (S.s).  
  For larger slider position, meaning faster effect speed, the transition time should reduce accordingly.  
  Math: t = 1000 -10 x S.s, where S.s = Segment.speed (scaled from 0.0-100.0), t = transition time in msec.  Ex: slider @ mid-position => t = 1000 -10 x 50 + 1000 = 500msec

If I were to make an official request, I would request all 3 options.

UI addition **[Palette Transition time: ◙None ○500ms ○Auto]**

Default WLED configuration automatically selects None.

El-Mare, 2020-09-11:
Hi,
I think what you look for can be found in LED preferences under transitions. The only negative effect I found is, that changing this value also affects the fading speed when the light is switched on / off.
I like @huggy-d1 's proposal a lot. May I friendly suggest a fourth option to set palette transition time and switch on / off fading time separately and not hard coded?

xxv, 2020-09-11:
The existing transition time doesn't apply to palette changes within the Random Cycle palette option. (I'm honestly not sure where it's used, TBH)

I poked at the code and it looks like I was misunderstanding how it works too: it randomly picks 4 colors and makes a palette from them instead of randomly choosing one of the pre-defined palettes.

El-Mare, 2020-09-11:
@xxv You are totally right! Sorry for confusing... Nevertheless, a fourth option, i. e. an at any time user configurable palette transition time, may still make sense for some users ;)

xxv, 2020-09-17:
Ah, whoops: I seemed to be getting "palette transitions" with preset transitions. Palette transitions work great with Random Cycle! Glad this is already in WLED.

closed 2020-09-17 by xxv

Aircoookie, 2020-09-17:
Yes, indeed, I believe Random Cycle only looks great with palette transitions enabled.
If you wondered, the reasons why it is off by default are that 
- it makes it harder to try out a lot of palettes because you have to wait for the transition
- it only works with one segment. You are correct that making the transition works with all segments would require storing 2 palettes (current and target) per segment. That's 96 bytes, or about 1kB for all segments. Doesn't sound much, but nothing we can easily afford on the ESP8266. Could be added optionally though via compile flags, and enabled by default on ESP32.

xxv, 2020-09-17:
@Aircoookie Maybe the palette transition setting could be surfaced a little better? It feels more aesthetic than things like the number of LEDs, the current limiter, etc. I think it'd go very nicely on the palette page.

Aircoookie, 2020-09-20:
@xxv yep, I will consider that :) the main transition duration can be set in the main UI as well after all

ChrisGebhardt, 2022-12-19:
Random Palettes seem to be jumping rather than smoothly transitioning when it switches (currently every 5 seconds hardcoded, not based on any UI transition time)  Perhaps there have been regressions since this issue was opened. I'll try to dig into the code if I get time.

blazoncek, 2022-12-19:
The code was since completely rewritten and yes, the change is instantaneous.
No workaround or immediate fix available ATM.