wled/WLED #1172: Random Cycle: synchronize palette

xxv, 2020-09-11:
When using the Random Cycle palette in a configuration where one device syncs to others, each device chooses its own random palette. Ideally, instead, the primary device should choose one palette and all the receiving devices should use the same randomly-chosen palette.

This would change the behavior of Random Cycle as follows:

When Random Cycle is chosen from the palettes on a device with sync enabled, every time it picks a new palette it sends a palette change over the network. Each receiving device would then see the randomly-chosen palette instead of Random Cycle.

Alternatively, this functionality could be broken off into a separate palette called Random Cycle Sync to preserve existing functionality where each device chooses its own palette. I'd recommend against adding more options unless the existing functionality is particularly desired.

(I'm happy to lead development of this feature, I just figured I'd put this up for feedback/comment before putting up a PR)

dexn, 2022-08-19:
I just started using WLED for this effect being synced but came across this problem. I'm using analogue strip's and would have loved them to all show the same random colour.

KeimpeRauwerda, 2023-01-27:
I also would really like this feature. I got it to sync once on accident and now I don't ever want it to be unsynced.

DarthWeber, 2024-11-06:
Random Cycle is really nice but i also would love to have it synced to other instances.  Hope it will find a way in further releases!

netmindz, 2024-11-07:
Depending on your setup it might make sense to use a master controller and DDP rather than independent ESPs each generating their own effect. 

I think the sync code could do with a bit of an overhaul in general as it's quite fragmented and confusing, but that's quite a major job