wled/WLED #1173: WLED Integration over Serial

rcarreira88, 2020-09-11:
Hi all.

Is it possible to integrate WLED with Ambient Light Aplication over serial?
Website: http://ambilight.tender-complex.ru/en/
Android App: https://play.google.com/store/apps/details?id=com.sevson.androidambiapp

My objective is to have Ambilight connected to my Android Box to play content from multiple sources (Netflix, Kodi, ...) witout the need of extra parts (RPI, HDMI Splitter, ...)

saulchristie, 2020-09-21:
I agree this would be useful. I wanted to move my desktop setup away from Hyperion --(UDP)--> WLED on my PC to remove the Wifi dependence only to realise I couldn't run WLED on the Arduino to receive data over USB... Ended up having to deploy a second Arduino(!) with this sketch:

https://www.github.com/Benik3/Adalight_WS2812_ESP8266_Non-blocking/

It would be incredibly useful for WLED to accept simple serial Adalight commands on the USB interface given the breadth of input options already in place.

Aircoookie, 2020-09-21:
Adalight via USB Serial is supported by default in WLED, it should work with both your applications :)
Keep in mind the baud rate is 115200 which may be too low for lots of LEDs. In that case you would need to do a custom compilation with baud 230400.

saulchristie, 2020-09-21:
Hey thanks @aircookie. I just reflashed WLED_0.10.2_ESP8266.bin (which is what I thought my unit previously had) and it is indeed working. That is, Hyperion NG --(adalight at 115200bd)-->WLED is working perfectly.

I wonder if I had atuline's fork on that unit previously or something? Suffice to say as far as I'm concerned the adalight support is working and it would be good if OP could confirm if his problem is resolved by flashing the 'official' 0.10.2 build so anyone else stumbling into this in future knows whether his proposed Android set up works or not.

Thanks for the quick heads up - very much appreciates and now I don't have to run dual Arduinos! Saved $5. <thumbs>

nirkons, 2020-09-22:
@Aircoookie - 
Hi, Im switching from an Arduino Uno to an ESP32 with WLED but I'm using a baud rate of 1500000 with the arduino which is working great.
I had 2 questions
1. do you think I can still retain that baud rate with the esp32? if not what is the highest baud rate you recommend
2. i saw you commented on an old issue but I can't find where in the code exactly to change the baud rate and compile, also for some reason when I compile and flash firmware.bin from \travis_esp32 (even without changing any code) WLED becomes unstable, the AP goes on/off and the LEDs flash once every few seconds as if its rebooting

Aircoookie, 2020-09-22:
@nirkons it should work. If you run into issues though you could drop back to 921600 baud which is verified to work!
`travis_esp32` is the test environment with every feature and debug enabled - it is not recommended to use. Use the `esp32dev` environment instead, it should yield more stable results. (you can just un-comment the esp32dev default env in platformio.ini)

I assume you have already found the location where to adjust the baudrate, just in case though in current GitHub master `Serial.begin()` is located in line 153 of `wled.cpp` :)

nirkons, 2020-09-22:
> @nirkons it should work. If you run into issues though you could drop back to 921600 baud which is verified to work!
> `travis_esp32` is the test environment with every feature and debug enabled - it [...]

I eventually found it I wasn't sure if that was the only place I needed to change, thanks :)

I tried a few different baud rates including 921600 and they seem to work except for the fact that every 5-10 there is a quick bright white flicker of all the LED lights, regardless of whats on screen, im using the latest [Hyperbian](https://github.com/hyperion-project/HyperBian) with adalight config.

this doesn't happen when using WLED configuration as LED Hardware or when using the default 115200 esp32 firmware (but 115200 results in delays), WLED over wifi is better than serial 115200 but still has a bit of delay, any idea what could be causing that flicker?
for now I'll try reinstalling hyperbian and see maybe its related to that.

thanks

Aircoookie, 2020-09-23:
The flash/flicker is likely not caused by the serial transmission, but by LED addressing issues. You might be able to mitigate it a bit by lowering the FPS/refresh rate in the Hyperion config. It can happen if the refresh rate of the LEDs is too high (>60fps), but can also happen with lower refresh rates occasionally if the data signal voltage is too low to be reliably read by the LEDs. It is fixable, but requires either a diode or a level shifter: https://github.com/Aircoookie/WLED/wiki/FAQ#reason-2

nirkons, 2020-09-23:
> The flash/flicker is likely not caused by the serial transmission, but by LED addressing issues. You might be able to mitigate it a bit by lowering the FPS/refresh rate in the Hyperion config. It [...]

Yes you are right I am not using a levelshifter/diode, connecting it directly to the 3.3V logic, but I just double checked and with WLED configuration in hyperion its working without a single flicker and and also worked fine with the Arduino Uno
Works: No flicker at all
[image]

One flash every 5-10 seconds:
[image]

Hyperion visualization isn't showing the flicker so its not directly coming from it, but its something with the serial, is there anyway to debug this?

thanks

nirkons, 2020-09-28:
As an update,

I tried again using - 
A different ws8212b strip
A different NodeMCU
A Windows PC w/ Prismatik (vs the previous RPI4 + Hyperion)

The same intermittent flashing happens on any custom build with higher than 115200 baud rate unfortunately, is there anyway i can get debug data on whats going on?

thanks

closed 2021-02-06 by stale[bot]