wled/WLED #1174: CY=~ support for cycle toggling

lab4d, 2020-09-11:
Hi,
working on a LED lamp using this amazing library, and i couldnt find a way to start and stop the preset cycle function using the button. As of now i have the lamp boot up with cycling enabled, and i use a CY=0 macro on double press to stop it. But the only way to start it back up is to replug the lamp.
**TLDR: Allow the "CY=" request to accept the ~ value for toggling.**
Or alternatively, but likely way more complicated to execute: preset slot 16 could allow being a preset cycle, which would allow using the button to get to the automated cycling and out.
Thank you to Aircookie and everyone involved for this amazing code <3

pbolduc, 2020-09-14:
The `~` is problematic from an API design as the parameter would be defined as an integer. There could be a possibility of using `-1` to indicate to toggle the current setting.

```
CY=0   -> off
CY=1   -> on
CY=-1 -> toggle between on / off
```

This is only my opinion, I will leave it up to Aircoookie for his thoughts.

Aircoookie, 2020-09-20:
Hi and thank you for the suggestion!
Just added `CY=2` to provide this functionality (2 as a value is in line with how the T= command for main power toggle works)

commit 35b54e2f (2020-09-20): Add CY=2 for cycle toggling (closes #1174)
closed 2020-09-20 by Aircoookie