wled/WLED #1177: HTTPS support for HTTP request API

SpicerSolutions, 2020-09-12:
Is HTTPS likely to be support on the HTTP request endpoint?

I appreciate this might be difficult for people with expiring SSL certificates.

Aircoookie, 2020-09-12:
Hi!
This is not likely/almost impossible to implement on ESP8266 due to insufficient memory.
On the ESP32 it is a different story, HTTPS support is easily possible. Still we would need to figure out how to add a certificate that is ideally both not self-signed (so that it is trusted by browsers), but that could not be circumvented by the private key becoming public due to the open source of WLED. This would likely require users to generate/install their own certificates for true security, but it's definitely something I want to look into!

pbolduc, 2020-09-12:
@SpicerSolutions  You can hear `Unexpected Maker` talk about the issues with https on this video: https://youtu.be/A4Hzz84HuNY?t=733

Fonta, 2023-01-12:
Could use a selg-signed certificate by default. In the config you could then replace it with your own. That's how most devices work.

blazoncek, 2023-06-13:
This can be supported using reverse proxy which will handle HTTPS (and WSS) with recent modifications in 0.14.0-b3+.