wled/WLED #1180: Increasing number of segments causes UI TypeError

twlare, 2020-09-12:
**Describe the bug**
The user interface reports "TypeError: Cannot read property 'shift' of null" when more than 11 segments are defined.

**To Reproduce**
Increase MAX_NUM_SEGMENTS from 10 to 12 in FX.h. Compile and download the new code. Use "+ Add segment" until you get to segment 11. Save segment 11 and refresh the UI to see the error message. The same error occurs when defining segment 11 through the JSON interface. The error can be cleared by resetting the segments.

**Expected behavior**
If MAX_NUM_SEGMENTS can not exceed the default of 10, then there should be a note in FX.h.

**WLED version**
 - Board: NodeMCU v2
 - Version: 0.10.2
 - Format: self-compiled with change to FX.h

**Additional context**
The error message is visible on both desktop and phone browsers. 
It is possible to control segment 11 through the JSON interface, but not through the UI.

Aircoookie, 2020-09-13:
Hi!
The segment limit is indeed set to 10 because there is not enough memory to transmit more than that reliably via the `/json` endpoint (`/json/state` will work with a few more, maybe 13-14, because the info object is not included).
You are right, I shall add a note to FX.h to explain this limitation. It should also be possible to get an increased number of segments to work on ESP32 as we have a lot more memory for the `JsonDocument` on that platform, I will look into that.

twlare, 2020-09-14:
Thanks for the quick response. I have 16 segments working, but because of the UI being frozen I have to program them via JSON or custom code. Both of these methods work with 16 segments so it appears that only the UI has a problem, while the core WLED code works fine.

t3chguy, 2021-01-14:
I am running 16 segments (144 LEDs in a 4x4 grid 9 leds each) on an ESP8266 and it works perfectly fine via HomeAssistant but as expected the Web UI is flaking out. Clicking things like Sync sometimes unblocks things and gets it to work. Is there any way to increase limits possibly at the cost of disabling some features?

Thanks

Kadah, 2021-02-24:
Running in to similar errors trying to use 12 segments on an ESP8266 D1 mini.

Some better segment support in the future would be cool.

sansillusion, 2021-02-24:
As aircookie has stated this is due to lack of memory. Nothing we can do but remove features effect etc... I would recommend using an esp32 instead.

closed 2021-07-09 by stale[bot]