wled/WLED #1181: Static IP not working? And Wemos D1 Mini Data Pin ?

selcuktncr, 2020-09-14:
Hello, Firstly, sorry for my english.
i entered a static ip and saved it. But result not changed. need to enter static gateway ?

my Second Question is,
Which binary need for Wemos D1 Mini ? i flashed this WLED_0.10.2_ESP8266.bin. is it right? if so, default data pin is D4? and can i change ?

Thanks.

Aircoookie, 2020-09-14:
Hi,
yes, you will need to set the gateway to the IP address of your router!
That is the correct binary for your board, and yes, D4 is the default pin. It cannot be changed at run time at the moment, but there are also binaries available for using D2 (pin 4) or RX (pin 3) as data pins :)

selcuktncr, 2020-09-18:
thanks for your answer. i solved my issues 👍

closed 2020-09-18 by selcuktncr