wled/WLED #1187: E1.31 data seems to create a null or blank pixel at the start of each universe after the 1st universe

CaffeinatedFunctionality, 2020-09-16:
**Describe the bug**
E1.31 seems to be making a null or blank pixel at the 171st pixel and each 1st pixel for each universe after the first.

**To Reproduce**
I would expect you can show this on anything running e1.31 with multiple universes

**Expected behavior**
The pixels to light up

**WLED version**
 - Board: ESP32 dev
 - Version 0.10.2
 - Format compiled with platformio

Aircoookie, 2020-09-17:
Copying from forums, since I saw your post there first :)

Thank you for reporting, I could indeed reproduce the issue using xLights!

After 2hrs of troubleshooting, managed to fix the issue. It was an off-by-one error that actually caused the last LED of the first universe (170th) to stay unlit.
The strange part is that this didn’t happen with LedFX (my go-to E1.31 testing client for its simplicity), despite it being set up for 510 channels as well.
In any case, the issue is fixed now in latest master!

CaffeinatedFunctionality, 2020-09-23:
sweet thank you so much man! I appreciate this fix! I'm donating!

CaffeinatedFunctionality, 2020-10-02:
I can confirm this is fixed. I'm closing this issue

closed 2020-10-02 by CaffeinatedFunctionality