wled/WLED #1188: How to address individual segments in a custom animation

Dann-io, 2020-09-17:
Hello, 

I was entirely sure where to ask or find this information. My question applies to version 10 plus of WLED and it is about building custom animations. Preferably using segments. 

I am building signs using normal LED strips and would like to animate segments (words) 

[image]

I would like to light up each word and random flash them on and off. My problem is in looking through the code is that all animation are the current segment. What I would like to understand is how to address each segment independently in a custom animation.

ie.
Animation 1
  Word/segment  1, 3 ,5 on
  Word/segment 2,4,6 flashing on. then stay on for a bit.
Animation 2 ( variations on the above)

It seems similar to the clock overlay stuff, but not enough for me to make sense. 

I have about 30 different LED lighting things around the house and at friends and this would help me with more. 

Any information would be brilliant, Thanks.

Aircoookie, 2020-09-17:
Hi! Yes, that is correct, that functionality is not (yet) available in WLED by default.
There are two ways you can make it work by first saving each word as a segment:
1. Use an external device to switch segments on and off on time with API calls like `{"seg":[{"on":true},{"on":true},{"on":false}]}` (this example would turn on the first two segments and the 3rd off, so it would illuminate "Facta non")
2. Make a usermod that switches the segments on and off with the desired timing/animation. This can be done with e.g. `strip.getSegment(0).setOption(SEG_OPTION_ON, true)` 

I'm planning to make a "Playlist" feature soon which will be able to load a sequence of presets, each saving different segment configurations. That can also be utilized to make animations like this, but it is not implemented yet unfortunately.

Awesome sign by the way, looks amazing with the neon-style diffuser tubes! :)

Dann-io, 2020-09-20:
Thanks for the quick response. I'm sure I'll have more questions as I attempt probably option2. Is this the best way to get my questions answered? Or should I ask/post somewhere else? 

Also thanks for the sign feedback. The software is awesome!

Aircoookie, 2020-09-21:
Here is good! You can also join the Discord server or make a thread in the forum, links to both in the readme. Whatever you prefer :)

closed 2021-01-31 by stale[bot]