wled/WLED #1192: Can

Moskus, 2020-09-21:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

Aircoookie, 2020-09-21:
Hi!
They are saved by activating the preset cycle. Just turn it off, set them the way you'd like and turn the cycle on again :)

Moskus, 2020-09-21:
Thanks! 😊
Now I just feel really stupid... That was apparently the only thing I didn't try.

closed 2020-09-21 by Moskus

Aircoookie, 2020-09-21:
No worries, this is an error on my side :) There really is no reason why these inputs should reset if any different control command is given.

Morphy99, 2020-11-29:
@Aircoookie changing transition time from the favourite tab doesn't seem to save for me. I have to change it in LED Preferences.

Aircoookie, 2020-11-29:
@Morphy99 That setting is only temporary. After changing it, use the power button to apply it and that will be the transition time until the next reboot. To change the settng permanently, use LED preferences :)