wled/WLED #1194: Does WLED buffer Artnet/SACN

abyssdj, 2020-09-21:
The long and short of this question is as you'd expect. Does WLED buffer streaming input? I'm having an issue which seems to only occur when using multiple universes (even with a small amount of LEDs, 90 max on each) where my lights are getting more and more lagged. Then, on quitting the software (Lightjams, or ELM), the lights continue to "play" the programming for a fair few seconds before pausing and then reverting to auto. Has anyone else come across this? It's almost as though WLED can't keep up, yet there doesn't appear to be any noticable slowdown in terms of refresh rate.

huggy-d1, 2020-09-22:
A simple test to know where the update lag lies:
Run the system until the lag is quite noticeable.  Unplug the network connection for the device sending Artnet/SACN to WLED.
What happens to WLED?
Does it immediately stop updating?
Does it catch up and then stop?
Does it do something entirely different?

Now, the trick is to reconnect the network connection on the computer running the software, and observe what happens to Lightjams/ELM and WLED.

From these tests, you can see where the lag is generated.

Choices:  WLED, network infrastructure, computer network buffers, computer software, windoze.

abyssdj, 2020-09-22:
In my haste (so much going on at the moment) I'd not thought about this. I am expecting it to be my router, but we'll see. I'll keep this updated.

huggy-d1, 2020-09-22:
A fresh perspective sometimes helps.  It very well could be the router.  I doubt it since most routers do not buffer much network traffic in the way you describe - as being noticeably delayed.  That said, you may have a particularly keen ability to notice such things too.  Do others also notice the lag?

abyssdj, 2020-09-22:
Yeah, as I say, that much rattling around my head at the moment that it's a basic step I'd not thought about. Nobody else has really had the chance to notice, or probably would ever notice. I do, though! I'll do some tests tonight.

Aircoookie, 2020-09-22:
Just my 2 cents to add to this - WLED does not explicitely buffer the streaming input. It just wouldn't make sense as the latency is greatly increased. Rather, as soon as a packet is received it will be drawn to the LEDs and discarded to minimize latency.
However, it could be that the ESP WiFi stack buffers some packets until the user application fully processed the previous packet. Still, I also expect the router to be the unit doing the buffering here, an ESP - especially an 8266 - doesn't have enough memory to buffer seconds of fluid animation worth of packets. I've noticed this issue too in some testing with large LED counts and currently the only mitigation seems to be to lower the frame rate and/or universe count per controller. The maintainer of the E1.31 library I use recommended setting the clock frequency to 160Mhz (default is 80Mhz in WLED), that might be worth giving a try as well.
To really solve this, we would need some logic like: "If i'm sending/processing packet 1 right now, packet 2 is in the buffer and packet 3 just arrived, discard packet 2 and draw packet 3 straight away", but if it is indeed the router doing the buffering, it is unlikely that you could change that behavior somehow...

abyssdj, 2020-09-23:
Thanks for the info. I'll do some experimenting!

abyssdj, 2020-09-24:
After a bit of tinkering, I'm still not sure on this one! As everything over here is wifi I can't pull a cable, but sure enough, turning the wifi off does leave the lights running for a short period before they run out of data. I tried it using my phone as a mobile hotspot which seemed to make the issue more apparent, so I'm guessing this is down to the router buffering the stream?

It seems very odd that having only 180 led's mapped (30 + 30 in universe 1, 60 in universe 2, 60 in universe 3) seems to work just fine with no noticable lag, yet if I add even a single pixel in a fourth universe, the lag mounts up very quickly. Reducing the FPS in Lightjams (or ELM) seems to help a bit. I wouldn't have thought 180 led's was even nearly pushing the limits of wifi? It seems to be universe related, if that's a thing? I'll do some more testing.

abyssdj, 2020-10-17:
Since my last post, I've tried this on a totally different router and totally different computer with identical results. I seem to get lag or delay with anything over 25fps output with only around 330 leds mapped. What I probably forgot to mention is that this is using sACN multicasting! As soon as I switch to unicast for each universe (in this case, each separate fixture), which in my situation is not really ideal, I can crank right up to 120fps with no lag at all.

closed 2021-03-19 by stale[bot]