wled/WLED #1195: WLED (Offline) With Decent Wifi Signal

ilandllc, 2020-09-22:
I've set up my ESP8266 exactly as shown in Dr. Zzz's tutorial videos.  The node is under my rooftop and connected to a 60A power source.  The Wifi signal is decent where the board is located (~25Mbps) but it's struggling to stay online within the WLED app.  

I hauled my laptop up the ladder and connected the node to the USB port and the WLED app on my iPhone instantly showed it as connected and I could operate the lights.  As soon as I unplug the board from the USB port on the laptop the app shows the device as offline.  

I've tried several different boards but they all have the same issue.  They show (Offline) in the WLED app unless I climb on a ladder and use my laptops USB port to power the board.  

Any advice for this noob is greatly appreciated.  My only goal is to have some awesome lights for the kids of my neighborhood during this incredibly depressing year.  

Thank you - Eric in Colorado

huggy-d1, 2020-09-22:
How is the board powered?  
Is it a QuinLED DigUno with ESP8266/ESP32?  
If it is not, then the 0V/GND on the power supply and the 0V/GND powering the ESP8266/32 are not electrically connected (logical guess based on the problem description and testing already performed).

While everything is powered up, a test to confirm this is to measure voltage from 0V on the power supply (what is wired to the LED strip) and the GND on the ESP8266/ESP32.  It should read very close to 0V.  The 0V to 5Vdc to the ESP8266/ESP32 board should read 5Vdc no matter what ground connection chosen.  

If everything is powered off, you can measure the resistance between those same locations.  It should read very low (close to 0Ω).

ilandllc, 2020-09-22:
@huggy-d1 - I appreciate the response.  All of this is very new to me so I apologize for my ignorance when it comes to any of this.  

I do not have a QuinLED DigUno but if that's something you'd recommend I'd definitely purchase one.  

Thanks again for responding.

ilandllc, 2020-09-22:
@huggy-d1  - to answer your question about the power supply.  
I have it connected to a DC5V 60a power supply.  I have the V+ and V- from the power supply connected to the red and white cables on the LEDS (WS2812B).  I used some pin connectors to run from the three pin connection on the leds to the ESP8266 - red going to the V1n, white going to GND and green going to data.

ilandllc, 2020-09-22:
I also just noticed when I disconnect the data pin from the board the WLED app on my iPhone connects and no longer shows it as offline.

sansillusion, 2020-09-22:
I have had a similar issue. When I changed the ground pin on the esp it
started working. It seems that there is an issue if I use the ground pin on
the same side of the Vin pin but if I use the ground pin on the opposite
side it works flawlessly.

ilandllc, 2020-09-22:
It worked for about a minute and then went offline again.  However, that was a nice minute of excitement.  ha!

Aircoookie, 2020-09-23:
@ilandllc if it is offline when connecting the data pin and you are alsonot seeing your lights turn on orange, likely your ESP is not booting. Try https://github.com/Aircoookie/WLED/wiki/FAQ#once-i-attach-leds-i-cant-connect-to-the-esp-anymore

closed 2021-01-31 by stale[bot]