wled/WLED #1196: No longer detecting lights and lights won’t turn on

rynozei, 2020-09-22:
I tried plugging my lights into the wrong end of the led strip and now the app no longer detects my lights and there now is only one green dot at one end and a purplish dot at the other. I then retried plugging them in on the correct side and the same thing happens. I’ve already reinstalled the flasher and tried messing with setting but now like I said before the app won’t even detect them.

uSyzex, 2020-09-22:
Hi,

Your LED strip is probably dead, or at least one of the led on the strip. 

Skip led 1 and connect to LED 2 to see if it works. 

LED strip are sensible to ESD.

rynozei, 2020-09-22:
I’m not exactly sure what you mean by skip led 1 and how how to just connect to led 2

pille, 2020-10-31:
@rynozei: you can usually cut LED strips. if you destroyed your first LED, cut if off and connect to the next one, until it works.

closed 2021-06-04 by stale[bot]