wled/WLED #1197: Api Hex value not correctly working

sife14, 2020-09-22:
Hello,
I have a led stripe with a extra white channel.
If I set an colour in the web interface with white in it and copy the hex value and use it in the api, an other color appears. 

**Example:** 

1. Hex value copied from the web interface: `ff7161ff` (orange)
2. Use in the Api: `win&A=255&CL=Hff7161ff`
3. Color of the led stripes according to the web interface: `2761ffff` (blue). The stripe is also light up in a blue color.

It kind of **shifts the two f** in the beginning to the end.

To get the "real" `ff7161ff` (a orange with white in it), i currently use as a workaround in the Api: `win&A=255&CL=Hffff7161.`

huggy-d1, 2020-09-22:
Is your strip configured in WLED as RGB, RBG, BRG, BGR, GRB, or GBR?
Does the color hex code match up with the way your strip is configured in WLED?

Aircoookie, 2020-09-23:
Thank you for the report! The `CL` command supports either an RGB color with the standard `RRGGBB` format, but when using RGBW, the format is (contrary to the HEX input in the UI) `WWRRGGBB`. I might consider changing it to `RRGGBBWW` which would make more sense, but I do not know whether other integrations use the API with RGBW hex codes and would therefore break if I make that change...

sife14, 2020-09-23:
> Thank you for the report! The `CL` command supports either an RGB color with the standard `RRGGBB` format, but when using RGBW, the format is (contrary to the HEX input in the UI) `WWRRGGBB`.

Thanks for your reply. It solved my problem. 

> I might consider changing it to `RRGGBBWW` which would make more sense, but I do not know whether other integrations use the API with RGBW hex codes and would therefore break if I make that change...

I think this would be a good idea, because it would add consistency through Wled, however I understand your "fear" of breaking integrations.

closed 2020-09-26 by sife14

huggy-d1, 2020-09-28:
@Aircoookie the current API is version 0.10.  When version 1.x comes out, you should consider providing the API version as an actual API input.

Then you can have these:
> <WLED_API_URL>?API=1&CL=RRGGBBWW
> <WLED_API_URL>?API=0&CL=WWRRGGBB