wled/WLED #1199: Constant exception after flashing latest master

dkneisz, 2020-09-23:
After flashing the latest master (state 23.09.2020) i get the following exception:

```
Exception (3):
epc1=0x4000df2f epc2=0x00000000 epc3=0x00000000 excvaddr=0x40280170 depc=0x00000000

>>>stack>>>

ctx: sys
sp: 3fffebc0 end: 3fffffb0 offset: 0190
3fffed50:  3ffea57c 00000000 00000000 00000000
3fffed60:  4025fcfc 3fff3504 00000000 4025fca8
3fffed70:  00000000 00000007 00000001 40280170
3fffed80:  00000002 00000000 00000000 3fff366c
3fffed90:  3fff2ea4 00000587 00000007 00481afd
3fffeda0:  3ffea540 2c9f0300 4000050c 3fff3504
3fffedb0:  00000007 3fff0f6c 00000007 402236dd  
3fffedc0:  40280170 00000000 00000010 ffffffff
3fffedd0:  4025f1c9 00000000 3fff2ea4 3fff0f6c
3fffede0:  00000002 00000039 00000007 40215e79
3fffedf0:  00003710 00040029 00070004 0006000e  
3fffee00:  e604003c 4bc6a704 f2b020c4 00000000
3fffee10:  3ffefe8c 0000000e 00000004 00000004
3fffee20:  00000006 00000012 40100690 00001275  
3fffee30:  00000000 00000000 00000001 3fff0a7c
3fffee40:  3fff3504 3fff0a84 3fff0f6c 40215ee4
3fffee50:  40100000 3fffc200 00000022 40100145
3fffee60:  3fff3504 3fff0a84 3fff0f6c 40223891  
3fffee70:  3fff3504 3fff0a84 3fff0a80 402238a4
3fffee80:  3fff0a94 3fff2ebc 3fff0a80 40264209
3fffee90:  00000001 7900a8c0 00000012 00000000  
3fffeea0:  00000002 00000012 3ffe8e1c 00000000
3fffeeb0:  40250000 00000000 00000001 3fff0c5c
3fffeec0:  3ffec51a 3fff0c64 3fff2ebc 40262b8d  
3fffeed0:  3fff09a0 3fff344c 3fff344c 40100598
3fffeee0:  00000008 3fff3084 3ffed728 3fff344c
3fffeef0:  3ffec50c 00000000 3fff2ebc 402620cd
3fffef00:  0200a8c0 0200a8c0 3fff0458 401005b9  
3fffef10:  3ffebee4 3ffebeee 3fff3450 00000002
3fffef20:  00000001 3ffeefa8 3ffeefd0 00481b6d
3fffef30:  40258662 3fffdab0 00000000 3fff34bc  
3fffef40:  3fffdc80 00000000 3fff2ebc 4025825b
3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49
3fffef60:  40000e19 00000005 00089c90 00000000
3fffef70:  3fffefc0 aa55aa55 000000ed 40104d59  
3fffef80:  40104d5f 00089c90 00000000 3d325200
3fffef90:  4010000d 3d324200 3d325700 3d584600
3fffefa0:  402510ac 3fffef3c 4025105d 3fffff58  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe
3fffefc0:  feefeffe feefeffe feefeffe feefeffe
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
3fffefe0:  feefeffe feefeffe feefeffe feefeffe
3fffeff0:  feefeffe feefeffe feefeffe feefeffe
3ffff000:  feefeffe feefeffe feefeffe feefeffe
3ffff010:  feefeffe feefeffe feefeffe feefeffe  
3ffff020:  feefeffe feefeffe feefeffe feefeffe
3ffff030:  feefeffe feefeffe feefeffe feefeffe
3ffff040:  feefeffe feefeffe feefeffe feefeffe  
3ffff050:  feefeffe feefeffe feefeffe feefeffe
3ffff060:  feefeffe feefeffe feefeffe feefeffe
3ffff070:  feefeffe feefeffe feefeffe feefeffe
3ffff080:  feefeffe feefeffe feefeffe feefeffe
3ffff090:  feefeffe feefeffe feefeffe feefeffe
3ffff0a0:  feefeffe feefeffe feefeffe feefeffe
3ffff0b0:  feefeffe feefeffe feefeffe feefeffe  
3ffff0c0:  feefeffe feefeffe feefeffe feefeffe
3ffff0d0:  feefeffe feefeffe feefeffe feefeffe
3ffff0e0:  feefeffe feefeffe feefeffe feefeffe
3ffff0f0:  feefeffe feefeffe feefeffe feefeffe  
3ffff100:  feefeffe feefeffe feefeffe feefeffe
[... 229 lines cut ...]
3fffff60:  3fffdad0 3fff03f8 402331a4 3fffefa0
3fffff70:  3fff0458 3fffdad0 3fff03f8 40233806
3fffff80:  3fffdad0 00000000 3ffef5da 4021a5ed  
3fffff90:  3fffdad0 00000000 3fff03b8 4021af60
3fffffa0:  feefeffe feefeffe feefeffe 402332c8
<<<stack<<<

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3584, room 16
tail 0
chksum 0xb0
csum 0xb0
v2843a5ac
~ld
Ada
```
I'm have this isue on a D1 mini and on 2 NodeMCU v2. This exception causes the controllers to reboot constantly. When I flash an older version (from 15.09.2020) it is working again, but all my presets are gone. All presets show up in the list, as if they were there, but when I click on them I get "WLED error 12". In serial monitor it says:

```
Read from /presets.json with key "1": >>>
null
```

Aircoookie, 2020-09-23:
Hi!
Please double check you are actually running master :) The serial output you got is only possible in the filesystem branch I am currently working on, which is not currently in a usable state.

dkneisz, 2020-09-23:
> Hi!
> Please double check you are actually running master :) The serial output you got is only possible in the filesystem branch I am currently working on, which is not currently in a usable state.

Im pretty sure. I have to go back to commit https://github.com/Aircoookie/WLED/commit/ff55a305cd8444cc7a03f0c74b8ead330ef1838f for it to work again. Every commit after that creates the mentioned error for me.

dkneisz, 2020-09-23:
This is the decoded exception:

```0x4025fcfc in tcp_write at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/core/tcp_out.c:633
0x4025fca8 in tcp_write at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/core/tcp_out.c:586
0x40280170 in chip_v6_unset_chanfreq at ??:?
0x402236dd in AsyncClient::add(char const*, unsigned int, unsigned char) at .pio\libdeps\d1_mini_no_blynk_hue_ir\ESPAsyncTCP_ID305\src/ESPAsyncTCP.cpp:256
0x40280170 in chip_v6_unset_chanfreq at ??:?
0x401019a0 in pp_post at ??:?
0x40102537 in rcReachRetryLimit at ??:?
0x40215e79 in AsyncMqttClient::_onConnect(AsyncClient*) at wled00\src\dependencies\async-mqtt-client/AsyncMqttClient.cpp:328
0x40100690 in millis at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:188
0x40215ee4 in std::_Function_handler<void (void*, AsyncClient*), AsyncMqttClient::AsyncMqttClient()::{lambda(void*, AsyncClient*)#1}>::_M_invoke(std::_Any_data const&, void*, AsyncClient*) at c:\users\derda\.platformio\packages\toolchain-xtensa@2.40802.200502\xtensa-lx106-elf\include\c++\4.8.2/functional:2073
0x40100000 in _stext at ??:?
0x40100145 in std::function<void (void*, AsyncClient*)>::operator()(void*, AsyncClient*) const at c:\users\derda\.platformio\packages\toolchain-xtensa@2.40802.200502\xtensa-lx106-elf\include\c++\4.8.2/functional:2465
0x40223891 in AsyncClient::_connected(void*, long) at .pio\libdeps\d1_mini_no_blynk_hue_ir\ESPAsyncTCP_ID305\src/ESPAsyncTCP.cpp:316
0x402238a4 in AsyncClient::_s_connected(void*, void*, long) at .pio\libdeps\d1_mini_no_blynk_hue_ir\ESPAsyncTCP_ID305\src/ESPAsyncTCP.cpp:510
0x40264209 in tcp_process at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/core/tcp_in.c:732
 (inlined by) tcp_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/core/tcp_in.c:340
0x40100788 in pvPortMalloc at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/heap.cpp:271
0x40262b8d in ip_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/core/ipv4/ip.c:559
0x40100598 in ets_post at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:177
0x402620cd in ethernet_input at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/netif/etharp.c:1379
0x40252907 in pp_tx_idle_timeout at ??:?
0x4025287a in pp_tx_idle_timeout at ??:?
0x4025825b in ets_snprintf at ??:?
0x40104d59 in call_user_start_local at ??:?
0x40104d5f in call_user_start_local at ??:?
0x4010000d in call_user_start at ??:?
0x402510ac in cont_ret at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/cont.S:142
0x4025105d in cont_continue at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/cont.S:51
0x40253323 in pp_attach at ??:?
0x40253372 in pp_attach at ??:?
0x40100598 in ets_post at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:177
0x401019a0 in pp_post at ??:?
0x4025242f in ppTxPkt at ??:?
0x40253323 in pp_attach at ??:?
0x40253372 in pp_attach at ??:?
0x4025347e in pp_attach at ??:?
0x4025242f in ppTxPkt at ??:?
0x402459bf in ieee80211_output_pbuf at ??:?
0x40100e64 in malloc at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:552
0x401052a3 in wdt_feed at ??:?
0x40100e64 in malloc at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266\umm_malloc/umm_malloc.cpp:552
0x40104287 in lmacRecycleMPDU at ??:?
0x40104c53 in lmacRxDone at ??:?
0x40100788 in pvPortMalloc at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/heap.cpp:271
0x40261eac in etharp_query at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/netif/etharp.c:1139
0x40103b4b in lmacProcessTXStartData at ??:?
0x40103b48 in lmacProcessTXStartData at ??:?
0x40262018 in etharp_output at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/netif/etharp.c:999
0x40102aab in wDev_ProcessFiq at ??:?
0x40262dfc in ip_output_if_opt at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/core/ipv4/ip.c:781
0x401028fc in wDev_ProcessFiq at ??:?
0x401028fc in wDev_ProcessFiq at ??:?
0x402312b1 in run_scheduled_recurrent_functions() at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/Schedule.cpp:167
0x402331e6 in esp_yield_within_cont at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:112
 (inlined by) __yield at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:130
0x4025105d in cont_continue at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/cont.S:51
0x402601aa in tcp_output at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/core/tcp_out.c:990
0x4025ed71 in sys_timeout at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/core/timers.c:296
0x4025e554 in tcp_connect at /home/gauchard/dev/esp8266/esp8266/tools/sdk/lwip/src/core/tcp.c:763
0x4022389c in AsyncClient::_s_connected(void*, void*, long) at .pio\libdeps\d1_mini_no_blynk_hue_ir\ESPAsyncTCP_ID305\src/ESPAsyncTCP.cpp:508
0x402159fa in AsyncMqttClient::setMaxTopicLength(unsigned short) at wled00\src\dependencies\async-mqtt-client/AsyncMqttClient.cpp:78
0x40100690 in millis at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:188
[... cut ...]
 (inlined by) esp_yield_within_cont at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:111
 (inlined by) __esp_yield at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_main.cpp:117
0x40233806 in __delay at C:\users\derda\.platformio\packages\framework-arduinoespressif8266\cores\esp8266/core_esp8266_wiring.cpp:54
0x4021a5ed in WLED::loop() at wled00/wled.cpp:96
0x4021af60 in loop at D:/Dokumente/Develop/WLED/wled00/wled00.ino:21```

dkneisz, 2020-09-24:
I did some additional testing. It looks like the commit https://github.com/Aircoookie/WLED/commit/edbe8131ab4da66836db4b2fafaa238429cb222b is causing this issue (at least for me). I cherry picked all other commits after that and have no issues. If I cherry pick this commit, I get the error.

One thing I didn't mention explicitly: All my NodeMCUs and the D1 mini were updated, so they had WLED running on them. I did the update via OTA first, but after that I fashed them directly connected to my PC. I have not tried flashing a "virgin" NodeMCU yet.

fishbone-git, 2020-09-27:
I believe I may be seeing the same problem.  ESP32s are all fine, but all of the ESP826 D1 mini's are not.  No time today to get then down and out of the boxes to check the exact exception via USB. Currently, just a constant boot loop so unable to do anything OTA. I can see the onboard led flashing every reset.

fishbone-git, 2020-09-28:
Confirmed, exact same dump here, D1 mini only.  D1 ESP32s are doing fine.

Aircoookie, 2020-09-28:
Thank you for the decoded stacktrace and report! I believe I might have a hunch what is causing this bootloop, namely me overdoing the `F()` macro for removing strings from heap memory. Are you using MQTT?

dkneisz, 2020-09-29:
Yes, I'm using MQTT.

Aircoookie, 2020-09-29:
Should be fixed now in latest master. This happens when you try and overdo optimizations, I should have realized that it would cause problems in that context. Sorry for the trouble!

dkneisz, 2020-09-30:
> This happens when you try and overdo optimizations

That's what we call "verschlimmbessern" ;)

I can confirm this working on my D1 mini and my 2 NodeMCUs.

closed 2020-09-30 by dkneisz