wled/WLED #1201: IST time removed from time setting timezone :(

sujitrp, 2020-09-23:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

Aircoookie, 2020-09-24:
You are correct, the 4 most recently added timezones (IST, Sasketchewan, Australia north & south) are missing from the dropdown menu... sorry about that! (note that this is only a UI problem, so if you are updating from a build where you could select the timezone it should remain in place)
As a workaround, you can set the timezone to UTC and put `19800` in the offset field.
Will fix ASAP :)

Aircoookie, 2020-09-27:
This is now fixed in master and will be included in the next release :)

closed 2020-09-27 by Aircoookie

Murffyp, 2025-04-17:
AST (atlantic standard time; eastern canada, 1 hour behind EST ( 8PM EST = 9 pM AST)) also appears to be missing... unless its under a different code...