wled/WLED #1202: Suddenly a single LED is lit while powered of, ransom led issues downstream from there

bwag62, 2020-09-24:
New/first install, 4 days of flawless operation(surprised even myself) and suddenly tonight I noticed one individual LED lit blue while powered down. I then turned on the LED's with the WLED app an all the LED's downstream of that singular blue LED go randomly crazy. Everything up to that LED works as directed. Occasionally it will work corectly as a whole but not for long. The LED in question is just short of halfway through the almost 600 total LED's.

Im a big time NEWBIE!! I've just used the pre programmed effects. Any suggestions where to start??

About 62 linear feet(4 sections 16.4" each) WS1812B(5V injection at each joint)
ESP8266 Node MCU ESP-12E
20amp power supply
WLED v0.10.2

Not sure what else might be helpful? THANK YOU!!

StefanRu1, 2020-09-24:
Hi bwag62,

i am not sure but it could be that this LED which stays on is defect.
I had the same issue with some stripes.
I could find out that the problems start when the LED is getting hot.

So i repaired mine by having it on full brightness and white for 5 min to get hot. 
Then changing color and check which is the first LED which does not work. 
Then i replaced the LED.
I continued this procedure until i had replaced all hanging LEDs.

So i had to replace 3 LEDs for a 5 Meter stripe. After this my stripe is now working since weeks without problems.
Even when it was running for hours on full brightness.
So i think there are strips where some LEDs are not very good and they will get defect as soon as they get too hot.
But most of the LEDs seems to be ok. 

Best regards,
Stefan

bwag62, 2020-09-25:
Thanks for the input Stefan! I used the weatherproof IP65 strips so I cannot replace individual LEDs unfortunately. I did try a different new NodeMCU with the same program tonight and I have the same problem so I feel it cannot be a faulty MCU. I did notice tonight that the "power on" erratic LED issue is from the suspect LED downstream but not all the way to the end...its about 18' worth that are effected and the last 3' work properly. 

  I just can't understand how the "problem" LED can stay lit while powered down in the first place? Somehow that one little LED is being told to light up blue 24/7 and wreak havoc when the strip is powered up...scratching my head.....

Legsmaniac, 2020-09-25:
> I just can't understand how the "problem" LED can stay lit while powered down in the first place?

Each LED had it's own built-in IC that controls each LED.  If one IC develops a fault, that LED could display erratic behaviour of its own independent of any input.

IP65 LEDs can be replaced but it's a bit of a messy job. What you do is cut the problem LED out by cutting along the dotted track either side of the LED, leaving reach half of the solder pads. Then clean the protective silicone off the solder pads using a knife or hot soldering iron down to the solder pads. Cut a spare single LED off some spare strip and prepare in the same way. Introduce into the pre-prepared strip, taking care it's the right way around by observing the directional arrows and solder the pads to join in the new LED. Seal with new silicone and allow to dry.

StefanRu1, 2020-09-25:
Hi,

yes the behaviour is exactly how you describe it for a WS2812B.
The faulty LED will do crazy stuff and all LEDs after the faulty one will also be affected.
There is an alternative WS2813 where one LED is not affecting the subsequent once.
See: https://www.seeedstudio.com/blog/2019/02/14/ws2812b-vs-ws2813-addressable-rgb-led-strips-how-are-they-different/

It is possible to repair a IP65 but it is a bit sketch to remove the silikone.
Because of this i do not buy IP65 anymore.
The silicone covering is also not very good for the heat dissipation, which might cause earlier faulte LEDs because of overheating.

Good luck in replacing or cutting out the faulty LED.

bwag62, 2020-09-25:
Thank you for the input guys I'm going to try & dive into it this weekend. Bummer they didn't even last a week and I have to tear into it all.

THANKS AGAIN!!

bwag62, 2020-10-01:
Follow up. Tonight I got up on the ladder and cut out the bad LED and the one on either side of it. It was a pain to spice it in on the house but its done and fingers crossed...it's fixed! Thank you all for the input!!

closed 2021-02-06 by stale[bot]