wled/WLED #1204: how to connect without a router?

TYR23, 2020-09-24:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

Aircoookie, 2020-09-25:
Hi!
Your WLED installation will be accessible using the `WLED-AP` network as long as no WiFi network is configured in settings or the software fails to connect to the configured WiFi within 20 seconds :)

TYR23, 2020-09-25:
I wanted to say if my router does not work, but I need to change the mode

TYR23, 2020-09-25:
thanks, I understood everything

closed 2020-09-25 by TYR23