wled/WLED #1206: Add PIN assignment to Info-Page

ElwoodSC, 2020-09-25:
Hi,

i love WLED ever since i discovered it and have built 10+ LED instances in the meantime. 
For different reasons I did some individual PIN assignments and now have the issue when updating my WLED instances.
Would it be possible (and hopefully also helpful for others) to show the PIN assignment (LED/DATA/BUTTON) in the info-page?

Cheers, 
Flo

huggy-d1, 2020-09-25:
I really like this suggestion.  Or at least having it available from a json or http api request.

Aircoookie, 2020-09-27:
I believe it's a good idea as well! Soon the pins will be runtime definable though and we'll only need a single build for esp8266 and esp32 respectively! For now it is indeed accessable in the API `/json/info`, where you can find `"pin":[2]`. Only does the LED pin though, in case of 2-wire LEDs it will not work and it doesn't indicate the button pin. So still lots to be improved :)

tablatronix, 2020-10-08:
How would this work? I am looking into it, and NPB methods and the wrapper are compile time, so pin assignments would have to be locked to the method, or compilation would have to include multiple methods.

blazoncek, 2022-07-25:
Runtime pin (GPIO) assignment is possible since 0.12 and is stored in config file which is retained during OTA updates.
Closing.

closed 2022-07-25 by blazoncek