wled/WLED #1207: Scrolling Text Feature

gazoscalvertos, 2020-09-25:
HI,

Congrats on a brilliant project you have here.

I was wondering whether scrolling text will be implemented natively and maybe accept input from the api / MQTT?

I know there is the option of using the TPM2.net protocol and stream over UDP but there isn't a great deal of options out there yet to handle that. 

I'd like to send messages from Home Assistant plus handling the text natively eliminates any noise on the wifi network which could introduce jerkyness on the scroll effect.

Thanks!

blazoncek, 2022-07-25:
The code is ready in my fork and will be available in WLED as well once 2D support is merged.

blazoncek, 2022-10-10:
Will be available in 0.14

closed 2022-10-10 by blazoncek

vogler, 2022-10-10:
@blazoncek When will that be? Do you have some reference? Didn't find anything searching for 2D or text.

blazoncek, 2022-10-10:
Soon. :tm:

jneilliii, 2023-01-02:
so seems the 2d matrix support and scrolling text options are out now in beta form, but documentation isn't updated on how to use it yet? or at least I couldn't seem to find anything related in the API or http requests methods. will those be updated or is there any reference to where we can learn how to remote control? looking to add text support to the OctoPrint WLED plugin to display print time remaining, etc.

blazoncek, 2023-01-02:
Unfortunately it has been not.
Use segment name for the text. The limit is 32 ASCII characters.

`{"seg":{"id":0,"n":"Your text up to 32 characters"}}`