wled/WLED #1210: Control ESP01S builtin LED

sasukebinbin, 2020-09-27:
Hi,

I like this so much after compare with ESPHome Fastled. 

I use it with ESP01S and implement in Home assistant. 

My problem is no way to control the builtin LED on ESP01S.

It is always on. I hope there is an extra button I can use to turn on or off the builtin LED.  And it will be recognised in Home assistant as an entity. 

Thanks
Rex

Aircoookie, 2020-09-28:
Hi,
thank you!
Unfortunately the builtin LED uses the same pin (GPIO 2) as the LED data and therefore it is always on as turning it off would break the data output.
Here is some more info: https://github.com/Aircoookie/WLED/wiki/FAQ#how-to-turn-off-the-bright-blue-onboard-led

huggy-d1, 2020-09-28:
Handy black nail polish or electrical tape works.  Desoldering the on-board LED is also good if you simply want it off.
If you *need* on-board LED to be separately controllable by HA, you can modify the LED pin in WLED code, compile for the ESP01S, and finally, move the data pin wire to your new pin.
If you want a button input, WLED already has a BTNPIN option.  It also has a RELAYPIN option.
With the BTNPIN, you can wire up a pushbutton that I believe controls RELAYPIN output, which you can define to be the on-board LED pin.
This may not work exactly as described, but perhaps by looking at the wiki (linked above by @Aircoookie, you may find other details that might work for your specific needs.

Aircoookie, 2020-10-12:
The builtin LED is now turned off when the addressable LEDs are turned off thanks to #1245 :)

closed 2020-10-12 by Aircoookie