wled/WLED #1211: Ультразвуковой сенсор HC-SR04

DivanX10, 2020-09-28:
Good day. I implemented zonal lighting in the kitchen. You can view it here https://youtu.be/XQdQc8rYKdA and here https://youtu.be/cMkz4YwIMG4. I use two controllers, one with Aircoookie WLED firmware and the second with EspEasy firmware for the HC-SR04 ultrasonic sensor. The Aircoookie WLED firmware has a usermod folder and I wanted to add the HC-SR04 ultrasonic sensor to it. I'm not good at code and I would like to have a ready-made plugin in the usermod folder so that I can get values from the rangefinder in the Aircoookie WLED firmware itself and configure which segments to use in it. Please add the HC-SR04 ultrasonic sensor to usermod and make it appear in the interface.

huggy-d1, 2020-09-28:
Thanks for sharing your videos.  I enjoyed seeing the vision and what you had working so far.

There are several libraries available that do the hard work for you:
https://tasmota.github.io/docs/HC-SR04/
https://github.com/fderrico/HC_SR04
https://github.com/sparkfun/HC-SR04_UltrasonicSensor
https://github.com/JRodrigoTech/Ultrasonic-HC-SR04

However, the majority of them seem to be branches of the same source library.
For use in a WLED usermod, I was looking at this one: https://github.com/fderrico/HC_SR04

I think that library would help you with the delays you were seeing.

Did you consider 2 HC-SR04's with one at each end of the counter?  Using slightly different field of views for each sensor to help discriminate small movements.

Reading the HC-SR04 once every 100ms (10 measurements / sec) would greatly reduce the lag.

Using the HC-SR04 with WLED would require using interrupt driven measurements - which the library I was looking at supports - instead of polling.  Polling would kill WLED's responsiveness and cause problems maintaining a WiFi connection.

That said, if you have some default settings, I think I could leverage existing WLED V2 usermods to create an HC-SR04 for you to test.

DivanX10, 2020-09-28:
Thank you for responding. I have two NodeMCU v3 controllers on a CH340 chip. On one controller is Aircoookie WLAD, and on the second controller is ESPEasy. I eliminated Tasmota after it failed to do what I needed. Have tassoti incorrect ultrasonic transducer. The tomatoes web panel displays the distance, but if you enable transmission to the MQTT server, the values are not transmitted. On ESPEasy, it works perfectly. I made the diodes offset incorrectly, I used &S= and &S2= and when the value changed to 1, MajorDoMo automatically shifted from &S=0 and &S2=40 to &S=1 and &S2=41, then &S=2 and &S2=42, then &S=3 and &S2=43 and so on, this created movement. I gave up on this idea and started writing a script in python to control effects and data is transmitted via JSON. JSON management works better and faster. While learning to write different scenarios for zonal lighting.

You wrote that if the same NodeMCU v3 controller controls the diodes and the rangefinder, this can lead to delays. But to understand this, you need to try. This option suits me better, but I don't understand how to add it to usermod. Here https://github.com/Aircoookie/WLED/wiki/Add-own-functionality it says that you need to put the code in a file usermod.cpp. Did I understand correctly that the HC-SR04 folder can simply be put in usermod, build the firmware and flash NodeMCU v3? If I understood it wrong, how do I do it? I would like to see the values from the ultrasonic sensor in the user Interface section of Aircoookie, and link its values to segments. For example, in the segments section, an additional item will appear where the distance will be specified and if the rangefinder value matches the specified range, this segment will be activated with the specified settings or use it in macros.

huggy-d1, 2020-09-28:
I think I understand what you are looking for.  That is probably more than I can do at the moment.  Maybe I can get it started and someone else can make it better.

If rangefinder programming does not use interrupts, that rangefinder code would make WLED quit working like it is supposed to work.

DivanX10, 2020-09-28:
Thank you again. I understand you. Then I will use two controllers. One with Aircoookie WLED, and the other with ESPEasy.

closed 2022-10-21 by Aircoookie