wled/WLED #1212: Newly flashed ESP32 AP Not appearing, but lights show default boot display.

rclough, 2020-09-28:
Hi there, thanks for all the hard work you've put into the WLED project.

I recently purchased a 4 pack of ESP32 dev boards, [these specifically](https://www.amazon.com/gp/product/B086MLNH7N/). I plugged one into a bread board with some WS2812B strips attached to data, and attempted to get up and running. I originally attempted to flash with NodeMCU PyFlasher, and when this didn't work, I tried ESPHome flasher, and eventually moved to using esptool.py. But the AP never showed up.

Somewhere in between the above I also I discovered the steps described in the (previous issue documenting the needed extra steps for ESP32)[https://github.com/Aircoookie/WLED/issues/517#issuecomment-571333712]. Doing this, I was able to get a 2nd board working as expected with this series of commands:

```
esptool.py --chip esp32 -p /dev/cu.SLAB_USBtoUART erase_flash
esptool.py --chip esp32 -p /dev/cu.SLAB_USBtoUART write_flash 0x10000 ~/Downloads/WLED_0.10.2_ESP32_ledpin16.bin
esptool.py --chip esp32 -p /dev/cu.SLAB_USBtoUART write_flash 0x00000 ~/Downloads/esp32_bootloader.bin
```

But the original board I attempted does not work with this, nor seemingly any order of the commands. What is strange is it does seem to be booting WLED: The first whatever handful of LEDs turn orange, and the buttons on the board change color/turn them on/off. But the AP is never displayed.

Is it possible this particular board has a defective WIFI module, or maybe I could have loosened a component when utting it into the bread board? Is there anything I can do to debug what may be happening?

Thanks for any help/direction! I'd rather not have to order extras if I don't need to

huggy-d1, 2020-09-28:
On board #1 (the problem child), does the erase_flash command stop the LEDs from doing anything after it completes?
If so, there is indeed hope, which there should be anyhow.

rclough, 2020-09-28:
@huggy-d1 IIRC, the LEDS would stay whatever color they were last set at as long as I didn't disconnect power to reset it (the power is coming from my power supply, not the ESP32, though they are both connected to the PS via the bread board power rail), but I'll need to try again later to be 100% sure of that. I tried a lot of things and may be misremembering

huggy-d1, 2020-09-28:
If you connect the ESP32 RX & TX & GND lines to a USB FTDI adapter's TX/RX/GND lines (swapping RX & TX), you can then use a terminal program to read debug data written by WLED.

rclough, 2020-09-28:
Is there a way to do this with just the existing micro USB connection or do i need to get some extra piece of hardware (FTDI adapter)?

toto79, 2020-09-28:
With this kind of esp32 dev Boards it should be passable to use your usb cable. Just plug it in and use monitor in vscode=platformio or serial monitor in arduino ide

jtheg777, 2020-09-28:
Does the blue led come on?
I've had those boards and when flashing I had to flash the bin file and a bootloader with this 
http://iot-bits.com/esp32/esp32-flash-download-tool-tutorial/

jtheg777, 2020-09-28:
Cant remember where I got the bootloader bin file from though.

rclough, 2020-09-28:
They package the bootloader in the assets for each release and theres also a link in one of the linked issues in my comment, but per my opening comment, I am flashing the bootloader, specifically this line in my original post:

```
esptool.py --chip esp32 -p /dev/cu.SLAB_USBtoUART write_flash 0x00000 ~/Downloads/esp32_bootloader.bin
```

I could try to do it with this tool though

jtheg777, 2020-09-28:
When I flash with Arduino IDE sometimes it doesn't startup and the blue led doesn't come on. I then save the bin file and flash with the above tool and the blue led lights and I can get to the AP.?

rclough, 2020-09-28:
Alright I'm able to do some more testing now. @huggy-d1 to answer your first question, erasing the flash does not change the LED colors that it booted with.

To go into a bit more detail, I just did this:

1. Plugged power into the board. The LED strip lights up the first  30(?, I didnt count) LEDs. Alsso posibly worth noting, iss that there's a red LED that is constantly lit. I checked and the working board seems to do this too.
2. Ran the following:
```
$ esptool.py --chip esp32 -p /dev/cu.SLAB_USBtoUART erase_flash
esptool.py v2.8
Serial port /dev/cu.SLAB_USBtoUART
Connecting....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
WARNING: Detected crystal freq 41.01MHz is quite different to normalized freq 40MHz. Unsupported crystal in use?
Crystal is 40MHz
MAC: f0:08:d1:d1:9b:24
Uploading stub...
Running stub...
Stub running...
Erasing flash (this may take a while)...
Chip erase completed successfully in 6.9s
Hard resetting via RTS pin...
```
3. The LEDs are still lit 30 orange, though they havent flickered or swelled (it seems on boot when the firmware is "correct", the lights swell in, and it's not doing that)
4. disconnecting power from the LEDs kill the lights. Reconnecting power they do not light back up.
5. I write the WLED bin:
```
$ esptool.py --chip esp32 -p /dev/cu.SLAB_USBtoUART write_flash 0x10000 ~/Downloads/WLED_0.10.2_ESP32_ledpin16.bin
esptool.py v2.8
Serial port /dev/cu.SLAB_USBtoUART
Connecting.....
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: f0:08:d1:d1:9b:24
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 1111856 bytes to 639619...
Wrote 1111856 bytes (639619 compressed) at 0x00010000 in 56.5 seconds (effective 157.4 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
6. Nothing happening. I write the bootloader:
```
$ esptool.py --chip esp32 -p /dev/cu.SLAB_USBtoUART write_flash 0x00000 ~/Downloads/esp32_bootloader.bin
esptool.py v2.8
Serial port /dev/cu.SLAB_USBtoUART
Connecting......
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: f0:08:d1:d1:9b:24
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 65536 bytes to 12827...
Wrote 65536 bytes (12827 compressed) at 0x00000000 in 1.1 seconds (effective 463.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
```
7. The lights swell on as expected. 

But after this, I wait a minute, and the AP never shows up :( 

To your point, @toto79, it seems either there aren't debug logs, or more likely my development board isn't set up to write to serial over USB, as the arduino serial monitor has nothing on it. Not sure if its worth noting that I had to install some USB to UART Bridge VCP driver to get it to register as a serial device at all, so not sure if thats somehow related to that lack of functionality.

Not sure where to go from here unless I purchase a  "USB FTDI adapter"? :/

rclough, 2020-09-29:
And unfortunately the ESP Flash Download Tool is not available for mac, although I'm pretty certain I'm doing the equivalent of what that tool would do, especially considering the steps work for a different board.

toto79, 2020-09-29:
@rclough : As mentioned in product description you habe a CP2102 chip onboard, so this is what you are looking for with a usb to uart bridge. And if flashing is possible, monitoring is also possible. Once you plug in the usb cable you should see an additional com port, which you have to select in arduino ide. Beside this you have to add some code lines to enable debug in wled in plattfomio.ini in the env section of your board.
 
`build_type = debug`
`build_flags = ${common.debug_flags}`

rclough, 2020-09-29:
Gotcha, the build flags I think were the missing piece, it wasn't clear that I should be building/deploying WLED to enable logging. I'll have to try and figure out how to get platformio working tonight, I've never used it before.

toto79, 2020-09-29:
you can give https://gitpod.io#https://github.com/Aircoookie/WLED/tree/master an try

1) pip3 install -U platformio
2) add Debug flags in [env:esp32dev] section
`build_type = debug`
`build_flags = ${common.debug_flags}`
3) platformio run -e env:type , e.g. platformio run -e esp32dev (as defined in platformio.ini)

find your firmware.bin file in folder .pio/build/env:type, save as... and flash to your esp like you did it before
good summary you'll find also at https://tasmota.github.io/docs/Gitpod/

salopette, 2020-10-12:
When I freshly flash the esp32 bin file, I don't see any AP from it.

Can someone upload a working bin?

RetroRule, 2020-10-17:
> When I freshly flash the esp32 bin file, I don't see any AP from it.
> 
> Can someone upload a working bin?

I seem to have the same problem flashing tasmota32.bin 
After a successful flash nothing happens. I tried another esp32 and I get the same result... nothing. Just a red led and no AP.

kernowquack, 2020-11-17:
I seem to have the same problem on D1 mini ESP32 clone.
Below is output from espytool about chip and  Arduino IDE serial monitor after erasing flash, flashing WLED_0.10.2_ESP32.bin and flashing esp32_bootloader.bin and restarting ESP32.

Have tried with 3 different boards of same type, will try with some other ESP32 boards.

**ESPTOOL output**

Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: 10:52:1c:69:fc:ac
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 65536 bytes to 12827...
Wrote 65536 bytes (12827 compressed) at 0x00000000 in 1.2 seconds (effective 436.9 kbit/s)...
Hash of data verified.
___________________________________________________________
**Serial monitor output**

entry 0x400806ac
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
ets Jun  8 2016 00:22:57

kernowquack, 2020-11-18:
@rclough @salopette @RetroRule
Don;t know if you guys are still having the same problem but I managed to compile and upload successfully using PlatformIo.
I then downloaded a copy of the binary which I then uploaded to 2 different ESP32 boards using the esptool.py program, I have attached the zipped bin file if that helps you out.
[wledEsp32_dump.zip](https://github.com/Aircoookie/WLED/files/5557077/wledEsp32_dump.zip)

WerkAG, 2020-11-18:
Many thanks kernowquack.
Regarding this matter of no AP with the ESP32 boards, I have tried all the suggestions I could read... nothings was worked. First time I tried with your bin file, Voilá!!! Bingo.
Tried with an ESP-32S board, and with an ESP32-ST, work on both.
Once again thank you, I was near to give up of the ESP32.

jlmeredith, 2020-12-03:
> @rclough @salopette @RetroRule
> Don;t know if you guys are still having the same problem but I managed to compile and upload successfully using PlatformIo.
> I then downloaded a copy of the binary which I then uploaded to 2 different ESP32 boards to test using the esptool.py program and it worked, I have attached the zipped bin file if that helps you out. Veersion10,0.2
> [wledEsp32_dump.zip](https://github.com/Aircoookie/WLED/files/5557077/wledEsp32_dump.zip)

I have successfully used your bin on my Adafruit ESP32 Huzzah (4MB).  I would like to know what you changed in the platform.io file when you compiled.  I have it set to both D1 Mini and ESP32dev in the config file and neither have worked for me.  Your bin did but I need to change the pin assignment and compile with that pin assignment.  I am building from master which is only 8 days old.

kernowquack, 2020-12-03:
I compiled for wemos_d1_mini32 using VSCode and PlatformIO, hope this helps. If it still doesn't work I will take a look and see if I can remember if I did anything else. 
Sorry, have a bit of a memory problem so not sure if I did anything else.

jlmeredith, 2020-12-03:
I determined the issue for my setup!  This maybe something to add to documentation or maybe it is there and I missed it!

The bootloader must be addressed to 0x00000 and the compiled firmware must be addressed to 0x10000 when running the esptool.py commands.  I have successfully compiled from esp32dev in the repo with a custom pin assignment and my changes are working as is WLED!  Woot!

kernowquack, 2020-12-03:
I remember now! Sorry had meant to write down what I did and upload but
can't find what I wrote. Sorry Guys.

Tony Hook
*Director*
[image: Steren-Limited]
t:  01736 759 140  e: tony@steren.co.uk
m:  07540 118 120  w: www.steren.co.uk

                                           [image: Find Us]
<https://hashpoint.org/gb/TR275DQ/B>                [image: Download
Directions] <https://drive.google.com/open?id=0B_IaavTnTneVazJFWmVxSl9QbG8>

Steren Ltd is a company registered in England and Wales. Registered number:
07853600. Registered office: Steren Ltd, 35 Treeve Lane, Connor Downs,
Hayle, TR27 5DQ. This message is private and confidential. If you have
received this message in error, please notify us and remove it from your
system. The opinions expressed in this email are the opinions of the
individual author and may not reflect the opinions of Steren Ltd

-- 

-- 

Steren Ltd is a company registered in England and 
Wales. Registered 
number: 07853600. Registered office: Steren Ltd, 35 
Treeve Lane, Connor 
Downs, Hayle, TR27 5DQ. 

This
 message is private and confidential. If 
you have received this message 
in error, please notify us and remove it 
from your system. 

The opinions expressed in this email are the opinions 
of the individual author and may not reflect the opinions of Steren Ltd

jlmeredith, 2020-12-12:
This is a big gap in both ESP32 documentation and WLED downstream documentation.  I have worked in tech related fields in the opensource community for 17+ years and it is always the small stuff that slips through the cracks, lol.  There are probably 20K people out there who tried to move to the ESP32 and this was a roadblock!  

My synopsis at this point is that the "tools" available for programming ESP based boards are all designed for a single register for uploading from a GUI.  The ESP boards introduce the ability to write you own bootloader, which requires a different address than the runtime code.  I am new to hardware level software so forgive any mistakes or lack of understanding.

I am going to block off some time to see how I can get involved in the WLED community as I have specific needs that it meets perfectly for my projects. I will see what I can do to get involved in helping improve things like this.  

Thanks for the input on this old thread.

DeviousPenguin, 2020-12-15:
> And unfortunately the ESP Flash Download Tool is not available for mac, although I'm pretty certain I'm doing the equivalent of what that tool would do, especially considering the steps work for a different board.

Is you have homebrew installed on a mac, then try installing `esptool.py`, it seems to work OK for me, you just need to `brew install esptool`

madrose, 2021-06-12:
I am facing exactly the same issue and tried wledEsp32_dump.bin poster by @kernowquack and I tried using all the tools I can find (command and GUI.  WLED_0.12.0_ESP32.bin will not run on my board and I tried several.

> esptool.py v2.5.0
Found 1 serial ports
Serial port COM6
Connecting....
Detecting chip type... ESP32
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse
MAC: xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB

Problem started when tried to upgrade to WLED_0.12.0_ESP32 after that it kept giving error:

> [18:07:30]ets Jun  8 2016 00:22:57
[18:07:30]
[18:07:30]rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:07:30]configsip: 0, SPIWP:0xee
[18:07:30]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:07:30]mode:DIO, clock div:2
[18:07:30]load:0x3fff0018,len:4
[18:07:30]load:0x3fff001c,len:1044
[18:07:30]load:0x40078000,len:8896
[18:07:30]load:0x40080400,len:5828
[18:07:30]entry 0x400806ac
[18:07:30]
[18:07:30]Brownout detector was triggered
[18:07:30]
[18:07:30]ets Jun  8 2016 00:22:57
[18:07:30]
[18:07:30]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:07:30]configsip: 0, SPIWP:0xee
[18:07:30]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:07:30]mode:DIO, clock div:2
[18:07:30]load:0x3fff0018,len:4
[18:07:30]load:0x3fff001c,len:1044
[18:07:30]load:0x40078000,len:8896
[18:07:30]load:0x40080400,len:5828
[18:07:30]entry 0x400806ac
[18:08:29]ets Jun  8 2016 00:22:57
[18:08:29]
[18:08:29]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[18:08:29]configsip: 0, SPIWP:0xee
[18:08:29]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[18:08:29]mode:DIO, clock div:2
[18:08:29]load:0x3fff0018,len:4
[18:08:29]load:0x3fff001c,len:1044
[18:08:29]load:0x40078000,len:8896
[18:08:29]load:0x40080400,len:5828
[18:08:29]entry 0x400806ac
[18:08:30].pio\libdeps\custom32_LEDPIN_16\LittleFS_esp32\src\lfs.c:998:error: Corrupted dir pair at {0x0, 0x1}
[18:08:30]E (41) esp_littlefs: mount failed,  (-84)
[18:08:30]E (42) esp_littlefs: Failed to initialize LittleFS
[18:08:30]Ada
[18:08:32]
[18:08:32]Brownout detector was triggered

I managed to get the bootloader successfully installed using esptool.exe not esptool.py on windows machine using the same options for the command line. once that is installed I installed a sketch using esphome and then used their GUI to upload WLED_0.11.0_ESP32_ledpin16.bin successfully. other than I could not get the botched upgrade to reinstall correctly.

beveler, 2021-07-19:
Hi - I've been having the same problems and found a workaround that works for me - Use ESPhome-flasher >flash with Tasmota > hard reset > flash with WLED and the WLED-AP appears as expected :)

closed 2022-04-16 by Aircoookie

thecheekymonkey, 2022-09-09:
I know this is an olf thread but its very few and far between when i dig out the old of box of esp 8266 and esp32 to have a play and come across this problem with both esp8266 and esp32. really struggle to get anything higher than 8.5 stably on the 8266 and 12 (with inlcuded bootloader) via notmal flashing routes. tried all software and flashing and had multiple instances of AP not showing.

stumbled across

https://install.wled.me/

which lets me flash the latest via webrowser and have successfully flashed all my problematic devices upto wled 13.3!!