wled/WLED #1214: SUNRISE -- Cannot enable via single HTTP API / Macro

henrygab, 2020-09-28:
## **Describe the bug**

It's desired to have a single macro / HTTP API call for each of the following:
* enable the sunrise nightlight effect
* enable the sunset nightlight effect

## **To Reproduce**

1. Manually set the power off
2. Use HTTP API with macro named 'Sunrise / Sunset' (_see below_)
3. Verify sun rises over the next one minute (_good!_)
4. Manually set the power on
5. Use HTTP API with macro named 'Sunrise / Sunset'
6. Verify sun sets over the next one minute (_good!_)
7. _Now, try to get force a sunrise, even if the power is on, with single API call:_
8. Manually set the power on
9. Use HTTP API with macro named 'Force Sunrise'
10. Notice that the LEDs turn off ... and then stay off (sunrise fails to work)

## **Expected behavior**

A single HTTP API call should be available that will deterministically select SUNRISE or SUNSET exists.
SUNRISE/SUNSET should not be forced to depend on the current power state of the LEDs.

## **Root Cause**

The code parsing the HTTP API...
1. runs linearly, not based on order parameters are sent.
2. parsing of nightlight parameters `NL=` occurs before parsing of master power `&T=`
3. parsing of master power `&T=` explicitly sets `nightlightActive = false`

Therefore, it's impossible, in a single HTTP API call / single macro, to:
* both ensure the LED power is OFF and enable nightlight for sunrise
* both ensure the LED power is ON and enable nightlight for sunset

## **WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

## **Additional context**

The HTTP API allows enabling a sunrise/sunset nightlight effect using, as an example, the following parameters:
* `&NL=1` to set one-minute total sunrise time (for testing purposes)
* `&NT=255` to set target nightlight brightness to 255
* `&NF=3` to set nightlight mode 3 (sunrise/sunset)

The selection of whether the nightlight will go to sunrise or sunset mode depends on the **_current power setting_**.  Thus, it would make sense that, by also including the master power setting, sunrise or sunset could be effectively selected by setting that as the first parameter:
* `T=0` to set power off should be the very first parameter.

## **Macro List**
Name | Parameters 
-----|-----
Sunrise/Sunset | `&NL=1&NT=255&NF=3`
Power Off | `&T=0`
Power On | `&T=1`
Force Sunrise | `&T=0&NL=1&NT=255&NF=3`
Force Sunset | `&T=1&NL=1&NT=255&NF=3`

henrygab, 2020-09-28:
There are a few ways to fix this.

1. Move the code that handles `&T=` to run before the nightlight setting is run (small change, potential change to existing behavior)

2. Have nightlight handling set a boolean to true, and prevent handling of `&T=` from disabling nightlight if that variable is true (smallest code change, but adds even more state to an already complex system)

3. Add two more fade modes: `&NF=4` to explicitly request sunrise, and `&NF=5` to explicitly request sunset (cleanest solution, but more types of nightlight setting may need more testing / code review)

I can generate a PR for any one of these.  Is there a preference?

henrygab, 2020-09-29:
Note: You might think this can be worked around with a couple macros.  Maybe, but simple ones aren't enough.
There's some state related to starting brightness that prevents the following from reliably working:

Macro | definition | description
-----|-----|-----
1 | `&T=0` | Power off
2 | `&T=1` | Power on
3 | `&M=0&NL=1&NT=255&NF=3` | Force Sunrise (Power off, then nightlight)
4 | `&M=1&NL=1&NT=0&NF=3` | Force Sunset (Power on, then nightlight)

Maybe it would be worth rewriting the HTTP request / Macro parsing, so parameters are always parsed in the order they appear?

Aircoookie, 2020-10-02:
Hi, first of all, thank you for documenting this issue in such a detailed way and even making a PR with a fix, it is highly appreciated :)

When developing the Sunrise nightlight, I originally considered selecting sunrise/sunset from the target brightness. That's what I should probably have done, but I didn't like that the target brightness, while selectable in settings and the API, is not selectable in the main UI (this could also be a potential future enhancement!). So I assumed that when the LEDs are on, the user probably wants a sunset, and when they are off, a sunrise is desired.

Allowing to force a mode makes sense though. I've implemented your proposed fix 1, handling the `T` command before `NL`.
I do not regard it as an unclean solution as the behavior is clearly defined by the command parsing order.
Also added the possibility to force a certain mode in the JSON API by combining the `on` and `nl.on` properties.

I believe rewriting the parser to handle the commands in the order given would not be that benefitial as it would 
- make the code more convoluted
- dissolve the defined order in which the parameters are parsed
- go against the way most APIs operate in that the order of properties given does not matter for the result. A good example is that, if order matters, `A=0&T=1` would lead to the exact opposite of `T=1&A=0`, even though you would expect them to do the same (currently, T is parsed later, so it has priority over A). This paradigm goes as far as that for JSON parsers and serializers, they are not even required by the standard to maintain the order of properties in a JSON object when manipulating it.

The only advantage I can think of would be, for example, allowing to save a preset without applying it to the LEDs with a call like `R=255&PS=1&R=0` which would save the full red value in the preset and then turn it off again immediately. Could also be used to save multiple, slightly different presets in one command though.

Despite the preliminary fix for API/macro usage I am still considering your PR for UI users who would like to force a mode! Then again, I regard the current auto mode selection to be quite intuitive for non-API use :)

henrygab, 2020-10-02:
Hi @Aircoookie ,

First, THANK YOU for adding the sunrise / sunset ability.  This is the single most useful effect, and I had been trying to figure out a good way to add this for some time.  AWESOME!

Thank you also for your detailed response...

1. **_You are right_** that adjusting when `T=` is handled allows deterministic selection via the HTTP API and macros.  Thanks for making that fix!

2. **_You were right_** to **_not_** use the target brightness as a method of selection.  Especially where used in dark areas, that heuristic would clearly have difficulty.

3. **_You are right_** about rewriting the parser.  I actually was a good way through trying to do that, when I realized that some systems / languages don't make it easy to specify parameter ordering when calling REST APIs, and that various parameters are interdependent.

4. During my attempt to rewrite the parser, I think I found some additional bugs.  _Look for more from me soon_.  😉

In closing, THANK YOU for WLED!

henrygab, 2020-10-02:
Closing as fixed by 3107943af81ae5b64267e83b36726d2e2e397f88.

closed 2020-10-02 by henrygab