wled/WLED #1216: Question on how LED Effects and palettes work and relate.

Boguinhos, 2020-09-29:
Hello all you !

I've been going thru the code to see if i could modify one effect, namely the sinelone_base, that is stolen from FASTLED example.
I've just found WLED today, so all my led implementations have been with custom code made by myself.
My two main and preferred effects are adapted from FASTLED examples, but i normally use hue color scheme for them, so i'm very lost on how these effects are built here on WLED.
All i wanted to is to remove the fade_out from sinelone and that each next led would take the next gradient color on the full hue scale. 
But i'm lost on the palette color range, intensity and segments, and the reason why that if i set my strip to 50 leds, commenting out the fade_out(SEGMENT.intensity); line i am able to make colors change without those led going black and i get somewhat of a close effect to what i need, but if i set the strip to 600, commenting it out still makes the led fade to black anyways.

[image]

here is the "video" example of what i wanted to achieve.
With fastled i am able to achieve it with this kind of code:

    if (cylon == String(1)) {
      time = millis();
      if (time-timeOld >50) {
if (z == NUM_LEDS || z >= NUM_LEDS){
  UP= 0;
}
if (z == 0 || z <= 0){
  UP=1;
}
                leds[z] = CHSV(hue++, 255, 255);
                if ( UP == 1){
                z++;
                }
                 if ( UP == 0){
                z--;
                }
                FastLED.show();
                timeOld = time;
        }
    }

Brightness is kind of ignored here as i only set the general brightness once and really don't feel the need to fiddle with it.

I'm not a pro coder or something like that and i could use some help. 
If i use color_wheel function as a replacement for my CHSV object i can get my "hue++" working but i can't get rid of the fading out of leds.
Can someone give me hand please? 
Thank you very much.

NeariX67, 2020-09-29:
You can find the fadeout option for the sinelon_base effect here https://github.com/Aircoookie/WLED/blob/master/wled00/FX.cpp in line 2595. Just comment that out and compile. You could also set the intensity slider all the way to the left, that should do the trick too.

Boguinhos, 2020-09-29:
> You can find the fadeout option for the sinelon_base effect here https://github.com/Aircoookie/WLED/blob/master/wled00/FX.cpp in line 2595. Just comment that out and compile. You could also set the intensity slider all the way to the left, that should do the trick too.

hi! thanks for your reply. That's exactly what i did that if i have set my leds to 50, it doesn't fade out. but if i set my strip lenght to 600, having it commented out doesn't make a difference! I don't understand why.

henrygab, 2020-10-06:
> The situation with that occurred by fading out the leds with 600 leds configuration was due to max brightness settings that automatically dimmed to black more distant leds from the actual led code position to save power.

If you have 600 LEDs connected, you are now into higher-power configurations.

600 WS2811 LEDs would take (approx.) 600 * 60ma ... 36A of power(!).  If true, STOP!  DO NOT TURN ON THE LEDS.  Seriously, you've given clues that you're creating a major fire risk.

Handling 36A of power is something you want to have an electrician review.  It's certainly not something you want to push through any ESP module.  For example, on AC, carrying that much current safely would require exceptionally thick wires (somewhere in the 6-8AWG range).  It may require larger wire for 36A DC current.

Lights at the end dimming?  That's more likely to be voltage drop.  That also points to insufficient current carrying capacity.

Terms for your search skills:
* LED "voltage drop" "power injection"
* NEC 2017 archive.org  ...  especially the tables on wire size for given current

BTW, if the power limiter is involved, it will dim *ALL* the LEDs equally, not just the ones at the end of the strip.

Legsmaniac, 2020-10-06:
> STOP! DO NOT TURN ON THE LEDS. Seriously, you've given clues that you're creating a major fire risk.

Hey, let's not scare people now with incorrect information.
Unless he's running ALL his 600 LEDs at white (all three colours red, green and blue on) and at 100% brightness, it's highly unlikely his 600 LEDs will draw 39 amps. Plus you would need to be using a 40A+ power supply to use 39A.  
Most effects use only a portion of the current, together with the fact that this is why we have the power and brightness limiter settings in WLED, from experience, most set ups are unlikely to draw more than a third of the potential maximum rated power.  So if he was using only a 10A power supply for example, he'll find most effects fine, possibly experiencing some dimming/flickering under more intense effects.  With a 10A power supply, it would therefore be impossible to draw 39A so any such setup would be quite safe.  😉

Boguinhos, 2020-10-08:
@henrygab and @Legsmaniac 
Thank you both for your input!

I've been using until before WLED about 10m of led strip x2 (600led, 60led/m)(one on ceiling and another at floor level)
For ceiling i've used 2x 10A power supply. One at the end and another at the beginning.
For floor, using 1x20A at end, 1x40A at middle and 1x12A at the beginning.
ESP8266 and custom code by me. Mostly for using the effect as pictured above. moving or solid. rarely for solid bright color.
FASTLED for power usage limiter.
2 years usage.
Never felt they were warm or drawing that much power.

Now with WLED i will have 2 walls of 10m in a room and in another room about the same setup as before, 2x 10m (ceiling and floor) but this time i will have 3x10A power supplies bridged at the middle of the strip but i will also run cabling (2x1mm2) to the other ends of the strip .

Biggest concern is with the 2x10m big straight walls in the other room as i only will be putting supply at the ends, i think i don't have space to run a wire to the 5m lenght.

I think it's not that much of a fire hazard. 
But i can be wrong! Inputs are always welcome on this matter.

Previous setup gave me confidence to step forward this way. As my ws2812b are not running very bright i would like to avoid putting them on a enclosure with light diffuser at front so that i can hide my parallel cabling. I would like them to "paint" a white wall with color and they will be on a ceiling with about 2.45 meters high. So i would not like to cut off any brightness at all for hiding cables.

 @henrygab "BTW, if the power limiter is involved, it will dim ALL the LEDs equally, not just the ones at the end of the strip." --yes, i've found this after the post. was not expecting the limiter to go fading to black, was expecting just less overall brightness as FASTED library does. And as my effect goes it behaved like fading to black sequentially. My poor judgment on this one. but sorted out. Been runing my custom WLED effect on a test 10 led strip for some days without problems, happy for being able to sort this out myself. It does not seem as nice as my original custom code, but i can live with this at the expense of WLED remaining capabilites! :)

henrygab, 2020-10-10:
I've read your response more than once, but I am still slightly confused.  Please allow me to focus questions, which should have short answers....

> ... before WLED about 10m of led strip x2 (600led, 60led/m)(one on ceiling and another at floor level)
> For ceiling i've used 2x 10A power supply. One at the end and another at the beginning.

OK, let me ask these questions, starting with the ceiling:
  600 LED == 36A (approx), and you used 2x 10A power supplies.

1. What did you limit power to in software?
2. Did you connect the grounds of the two power supplies, or keep them electrically separate?
3. Did you connect the positive voltage of the two power supplies, or keep them electrically separate?
4. How often and where did you connect power to the strips?
5. Did you run any cable for voltage injection?
6. Did you measure, for each electrically separated section ...
    6a. how much your ground shifted (from the first LED to the one furthest away)?
    6b. how much your voltage dropped (from the first LED to the one furthest away)?
    6c. how many amps the strip consumed?
7. Did you measure these values when the LEDs were at maximum brightness?
8. Did you install fuses?  (where and what amperage?)

> I think it's not that much of a fire hazard.

I can't say if it is or is not.   What I can say is that I prefer to avoid a fire hazard.  For me, this means:

A. Measurements of worst-case current (Amperage) ... all-white, full-bright
B. Calculation of minimum wire size required (requires Amperage, length of wire)
C. Calculation of correct fuse size
D. Calculation of necessary power supply (at 80% rated / oversized by 25%)
E. Measuring of actual voltage drop and ground shift (also at worst-case amperage)
F. Power injection as necessary (guideline: every 50x 5V LEDs, every 75x 12V resistor-based LEDs, every 100x 12V regulated LEDs)

At 600 LED, you have a potential for over 30A of current.  That's quite a bit.

Boguinhos, 2020-10-10:
Ok, i'll try. Thank for the help.

> OK, let me ask these questions, starting with the ceiling:
> 600 LED == 36A (approx), and you used 2x 10A power supplies.
> 
> 1. What did you limit power to in software?
10A via software. (fastled library)

> 2. Did you connect the grounds of the two power supplies, or keep them electrically separate?
grounds were merged thru the strip.

> 3. Did you connect the positive voltage of the two power supplies, or keep them electrically separate?
positives were connected thru the strip also.

> 4. How often and where did you connect power to the strips?
not sure i've understood. The power supply is connected to a dedicated circuit breaker of 220 16A

> 5. Did you run any cable for voltage injection?
no as i felt that a 10A power supply for 5m x2 could be enough.

> 6. Did you measure, for each electrically separated section ...
>    6a. how much your ground shifted (from the first LED to the one furthest away)?
did not measure. not sure how anyway with the DMM.
>    6b. how much your voltage dropped (from the first LED to the one furthest away)?
also did not measure.
>    6c. how many amps the strip consumed?
also did not measure.

> 7. Did you measure these values when the LEDs were at maximum brightness?
did not.

> 8. Did you install fuses?  (where and what amperage?)
No. I don't know how to do the proper calculations as it was never intended to go full white all the time.

> 
> > I think it's not that much of a fire hazard.
> 
> I can't say if it is or is not. What I can say is that I prefer to avoid a fire hazard. For me, this means:
> 
> A. Measurements of worst-case current (Amperage) ... all-white, full-bright
I know the math for (0,20mA x3 per led) x led amount). just that. i know how to measure Amperage with DMM but i was afraid to do it because i don't know how many amps can my UNI-T measure safely.

> B. Calculation of minimum wire size required (requires Amperage, length of wire)
i have chosen 2x 1mm2 wires because it is what is physically feasible for this installation for running power injection. i'm hoping that it helps and i will try to control the possible damage by decreasing maximum power consumption by WLED software.
also, thicker wires seem like impossible to solder with success.
  
> C. Calculation of correct fuse size
if you could help me somehow that would be great.

> D. Calculation of necessary power supply (at 80% rated / oversized by 25%)
again, i try to fit in whatever i can and then work my limit thru software.

> E. Measuring of actual voltage drop and ground shift (also at worst-case amperage)
> F. Power injection as necessary (guideline: every 50x 5V LEDs, every 75x 12V resistor-based LEDs, every 100x 12V regulated LEDs)
i will try to have at least in the middle of this strip. 

> 
> At 600 LED, you have a potential for over 30A of current. That's quite a bit.

thanks for you help.
And i'm no pro when it comes to electricity, as you may have noticed :) i know the basics and the formulas if i quick search google. i do understand the concepts. i just am not able to implement this project like it gonna look clean if i do all of the things exactly by the book. that's why i choose to have a power supply on each end of the strip.

thanks.

henrygab, 2020-10-14:
>    positives were connected thru the strip also.

While you were correct to merge the grounds, the positive voltage outputs should NOT be merged when using multiple power supplies.  You need to pre-determine how many of the LEDs will be driven by which power supply, and electrically separate the positive voltages from the different power supplies.  Better yet, only use one power supply.  Add more controllers, rather than have to deal with the complexity of multi-power-supply.

> ... _(removed lots of "did not measure" answers..."_ ...

If you want your project to work, you need to measure stuff.  **_Period._**  Measurements should be at worst-case load (100% bright, all-white, no limiter).  (Fuse first...)

**_Measure_**.  You are asking for analysis of your project, but refusing to measure the things that matter, and that can enable you to have a stable, successful result.  Most problems with LEDs in larger setups (in my experience) relate to voltage drop / ground drift.

>    no as i felt that a 10A power supply for 5m x2 could be enough.

Ensuring your power supply can provide the required amperage is one problem.  Note: plan for 80% maximum usage on any power supply.

An entirely separate problem is ensuring sufficient voltage.  Voltage drop is affected by load.  That is, without load, you're not going to see the voltage drop.

<details><summary>How to measure (click to expand)</summary><P/>

This is based on using a **_single_** 10A power supply for these 100 LEDs.

1. Lay out a loop of 100 LEDs (so the start and end are both near you)
    a. 100 LEDs should be ~6A, which is less than the 80% maximum load your power supply should run at...
2. Attach the power supply to the start of the strip, with a maximum 7.5A fuse
3. Set 100% brightness, and all 100 LEDs to white, to maximize the load.
4. Set your DMM to measure DC voltage
5. Measure voltage at start:
    a. Touch one probe (e.g., black) to the ground at the start of the loop
    b. Touch other probe (e.g., red) to the V+ at the start of the loop
6. Measure voltage at end:
    a. Touch one probe (e.g., black) to the ground at the **end** of the loop
    b. Touch other probe (e.g., red) to the V+ at the **end** of the loop
7. Measure voltage drop (V+ line of the strip):
    a. Touch one probe (e.g., black) to the V+ at the **start** of the loop
    b. Touch other probe (e.g., red) to the V+ at the **end** of the loop
8. Measure ground drift on the strip:
    a. Touch one probe (e.g., black) to the GND at the **start** of the loop
    b. Touch other probe (e.g., red) to the GND at the **end** of the loop

Fully made-up example results:
```
Q5 == +5.2V, Q6 == +2.8V, Q7 == -2.0V, Q8 == +0.4V
```
With those fake example results, if I then measure (by testing the specific strips I am using) the lowest voltage that is reliably usable on the strips as being +4.0V, I can then use math to determine the *maximum* number of LEDs before the voltage drops to an unstable level, and then subtract a percentage for other variables.

With +5.2V on input, and +4.0V minimum voltage target, I want to limit voltage drop to 1.2V.
100 LEDs == 2.2V drop, so approximately 0.022V drop per LED.   1.2V / 0.022V ~= 54.5

This means (approximately) 50 LEDs before voltage drops too much to remain stable under load.

</details>

> i know the basics and the formulas if i quick search google. i do understand the concepts. i just am not able to implement this project like it gonna look clean if i do all of the things exactly by the book. that's why i choose to have a power supply on each end of the strip.

If you want to reduce risk, you need to add fuses.  Period.  Devices and wiring fail.  Arcing and shorts cause fires.  What size fuse depends on the **_lowest_** rated segment.  So, if your strip is rated to carry 5A, you need 5A fuse.  If you have three power injection sites *AND* fuse each of those locations, they could each get a separate 5A fuse ... but only if you fuse your power supply output at 10A.

For stable power output, avoid exceeding 80% of most power supplies' rated output (e.g., plan for only 8A from your 10A power supply).  I recommend you avoid using more than one power supply per continuous LED strip for now.

If you don't do stuff "by the book", and you are (as you admit) a beginner in electrical, then you are taking risks.

PLEASE ... use fuses ... It could save your LED strip;  It could save your life.

Boguinhos, 2020-10-20:
@henrygab Hi again!
Thank you so much for your help, once again.
Will explain now:

> While you were correct to merge the grounds, the positive voltage outputs should NOT be merged when using multiple power supplies. You need to pre-determine how many of the LEDs will be driven by [...]

After reading your last post so i did change my strategy. Once upon a time i've should read something that led me to think that merging V+ was the way. For sure it was not my own idea after a good night of sleep because i was never too confidant to merge power supplies. As for GND, it's something that has to happen when using multiple eletronic devices, as you should know.
Now i'm separating my setup im max 5m lengths and each supply for that lenght. Single setup does not seem a good idea for me, besides setup difficulties i really want to run away from those huge Amp supplies.

> If you want your project to work, you need to measure stuff. **_Period._** Measurements should be at worst-case load (100% bright, all-white, no limiter). (Fuse first...)
>
> **_Measure_**. You are [...]

I didn't refuse, the thing that you might have not clearly understand is that some input that i gave here before relates to a previous setup that already is dismantled. And i'm still prepping up the new one, although i already have like 3 or 4 meters of strip laid down. Will measure those as soon as i have the chance.

> 
> > no as i felt that a 10A power supply for 5m x2 could be enough.
> 
> Ensuring your power supply can provide the required amperage is one problem. Note: plan for 80% maximum usage on any power supply.

I understand that. I just didn't want to throw away all those 7 or 8 10A supplies i have already. Knowing the math does not add up would it be really that bad to have 10A for less than 5 meters? And limit the power usage by software on WLED?
Now i have a more specific length for you measured up:
1x10A for 4m of strip.
1x10A for 3.5m of strip.
1x10A for 3.70m of strip.
for one section on the ceiling.
for floor it's about the same:
1x10A for 3.9m of strip times 3different walls.
All strips have at least 3x 1.0mm2 cables going into it. End, middle and start. 
I've fired 4m of strip up but just to check if each section that a "sew" together is working.

That's about 14Amp for 4m. i know the supply is on the low side.

For the other big room with the 10m wall i have purchased bigger supplies. 20A. That will go max for 5m.

> 
> An entirely separate problem is ensuring sufficient voltage. Voltage drop is affected by load. That is, without load, you're not going to see the voltage drop.
> 
> How to measure (click to expand)
> This is based on using a **_single_** 10A power supply for these 100 LEDs.

> 1. Lay out a loop of 100 LEDs (so the start and end are both near you)
>    a. 100 LEDs should be ~6A, which is less than the 80% maximum load your power supply should run at...
> 2. Attach the power [...]

> With those fake example results, if I then measure (by testing the specific strips I am using) the lowest voltage that is reliably usable on the strips as being +4.0V, I can then use math to [...]

I understood this. just have one question: I will always have to run cables to the strip from the supply. I would be better off to measure with cabling already connected to the start and end of the strip, at least?

> If you want to reduce risk, you need to add fuses. Period. Devices and wiring fail. Arcing and shorts cause fires. What size fuse depends on the **_lowest_** rated segment. So, if your strip is rated to carry 5A, you need 5A fuse. If you have three power injection sites _AND_ fuse each of those locations, they could each get a separate 5A fuse ... but **_only_** if you fuse your power supply output at 10A, AND everything else that might carry 10A is rated to do so (e.g., the power injection wire).

Ok, i have some doubts here. Lowest rated segment as the fuse has to be as strong as the weakest link on the chain. I think i got that. But how does this sound to you:
Power supply 10A ----(5 centimetres of power single wire 1.5mm2)> fuse ??? ( i would say 10A) ----> electrical splitter->>power injection wires (1.0mm2 multiwire , at least 3 independent cables for 4m max of led strip).
fusing each power injection wire seems like it's possible to blow one fuse on one injection wire and still having the other ones working, even if i put 10A divided by 3 fuses on each of the 3 cables. i think that one main fuse would seem more fit. And the nearest for the power supply as possible.

> For stable power output, avoid exceeding 80% of most power supplies' rated output (e.g., plan for only 8A from your 10A power supply). I recommend you avoid using more than one power supply per [...]

Will do. You are right. 
Time is a expensive resource and that's the least i could do for making you spend yours on helping me.

Any nice fuse holder and fuse type you recommend ?
Thank you once again.

henrygab, 2020-10-21:
First, I have to highlight that, while I can point out things that might **_NOT_** be safe that you may want to review, I'm neither qualified nor able to remotely state that your setup is safe.

Some sites I find helpful:
* https://www.littelfuse.com/products/fuses.aspx
* https://www.12voltplanet.co.uk/fuses-guide-uses.html

<details><summary>would it be really that bad to have 10A for less than 5 meters? And limit the power usage by software on WLED?</summary><P/>

Yes, it would be bad to rely on software limiters.  Use a 200A supply if you want ... just FUSE according to the lowest-rated component, and ensure all wiring can handle the fused current.  Don't rely on software, because the software isn't written to safety standards, and the MCU might be fried ... get a physical fuse in place as a fail-safe.

</details>

<details><summary>Measurements question...</summary><P/>

> Now i have a more specific length for you measured up:
> ... (4m max per 10A PS ... 4m may try to draw 14A ) ...
> All strips have at least 3x 1.0mm2 cables going into it. End, middle and start.
> ... (5m max per 20A PS ... 5m may try to draw 18A? ) ...

> > ...  voltage drop ... How to measure (...)

> I understood this. just have one question: I will always have to run cables to the strip from the supply. I would be better off to measure with cabling already connected to the start and end of the strip, at least?

Yes, start with the cabling already connected, if you don't want to manually calculate and adjust for any additional voltage drop / ground shift.

</details>

<details><summary>How I select fuses and design stuff...</summary><P/>

> But how does this sound to you:
> ... i think that one main fuse would seem more fit.  And the nearest for the power supply as possible.

Yes, if using one fuse, it should be as close to the power supply as reasonable.  In this case, every component must be rated to carry the full current of that fuse.  Without adding complexity, this is the only config.  **_Weakest link defines fuse size._**  After all, the purpose of the fuse is to protect the wires and components from carrying more current than they should.  Software cannot ensure this to any safety level.

Here's a hypothetical design I will generate, with explanation of what I choose.  As I'm not familiar with metric wiring amperage ratings (sorry!), so I'll use AWG, but mention what amperage it needs to carry (since that's the key point for fusing):

1. Power supply rating is 20A, but I would want it to not exceed 16A continuous.
2. Sizing the fuse for the power supply, I aim for 110%-150%, so I'm looking for a fuse in the 17.6A - 24A range.  20A is easily available, so I use that for the power supply fuse.
3. I want to use the same wire gauge everywhere, so I choose wire that's able to handle AT LEAST the current allowed by the largest fuse (in this case, 20A).  At 86 degree (F) ambient, with <= 3 wires running together,  I might select 16 AWG THHN/THHW wire.  With consultation and approval of a electrical engineer / electrician, I could potentially use thinner wire.
4. In this example, the LED strip rated for 4A (much lower than the fuse at the power supply).  To keep the LED strip from carrying more than 4A, I'll cut the V+ and V- (GND) every 50 LEDs (or 60, if I want to push the envelope).
5. For each segment, I connect from the segment directly to the main V- line.  Note that segment grounds are NOT electrically connected to each other, except through the main V- wire.
6. For each segment, I connect from the segment, via a fuse, to the main V+ line.  These fuses are rated lower, to limit the current the segment carries.  Since the strip is rated 4A, 110%-150% gives a fuse in the 4.4A-6A range; I select a 5A fuse for each segment.

Other random bits:
* I sometimes prefer to connect V+ to one end of each segment, and V- to the opposite end of the segment.  Note that this has more to do with keeping voltage drop approximately the same across the segment.
* I sometimes prefer to alternate whether V+ or V- connect near the start of the segment ... this depends on whether it's easier to splice two connections from the main V+ / V- lines at once, or easier to have more splices with only one connection. YMMV.

In the diagram, I use both these (entirely optional) random bits.

<details><summary>Why don't I leave ground connected between segments?</summary><P/>

Consider where the segments have ground directly connected, and then one of the segments' wire to the main V- line breaks.  Because the V+ line is still attached and there's still a ground connection, the LEDs can still draw ~4A.  Unfortunately, that current has to travel along another segment, which also might be drawing 4A.  That means a segment could carry double what you intended, without burning a fuse.  If the grounds are not directly connected, then the segment will just stop working.  This is better than doubling the current carried by a neighboring segment.

</details>

Again, **_only the data line_** will still be directly connected along the entire length of the strip.  In this example, I end up with four LED strip segments where the V+ and V- lines are electrically separated, with approximately equal quantity of LEDs per segment.

</details>

```
V- ----+-------------+-+-------------+
       |             | |             |
       [SEG 1] [SEG 2] [SEG 3] [SEG 4]
             F F             F F
             | |             | |
V+ ---F------+-+-------------+-+
```
[... cut ...]
I like automotive (ATC/ATO) fuses.  They are cheap, readily available everywhere, handle 12V, and it's easy to see if the fuse is blown.  In a more space-constrained environment, I might look at inline glass fuses, because they can sometimes add just a bit more bulk than the wiring itself.  An intermediate solution would be a mini-ATC or micro-ATC.

However, what you choose should match what you need it to do, and what's important to you.  LittleFuse carries many types, and includes simple descriptions and I've found their site helpful.  Buy fuses from local retailers or trusted electronic component sites ... don't save pennies on safety equipment.

</details>

Hope this helps...

Boguinhos, 2020-10-31:
Hi @henrygab 
I’ve been busy and slowly continuing the led project.
For 37 led lights ( odd number but it has to do with the strip division on the shortest lenght. I divided the total lenght in 4 sections:
My dmm measurements
4.758 start of section
4.650 end of section
0.0546 vdrop
0.0539 gnd drift

I fused all the sections and the master connection to the power supply.
The shortest fuse i had was 2A and i think it is high for the 37 led section. Will try a 1A one when i can. Master fuse i’ve tested with 5A (as proof of concept) and it slowly bursted under maximum load, 7.5 seems ok and within range. Wire temp seemed ok too on all sections. 
For now i’m keeping the multiple power supply (3x10) and i have to have the gnd merged otherwise the led strip wont light up on the section that is powered by the next power supply. What i did was to run power directly to the esp on a different wire, also fused, and split that gnd for all power supplies. I think it is not the best situation, but i can’t think on a better option. The strip sections are not connected together on the strip itself. Only after fusing on the wiring.

Power supply > + master fuse+ > + distributor > +section fuse+> section independant wire 1mm2 > strip+ > gnd wire back > gnd distributor > power supply -

This for all sections.

How am i coming?
I havent cable tidy this but i was testing and took a picture:
[image]

Thank you for all your help again

Boguinhos, 2020-10-31:
[image]

[image]

henrygab, 2020-11-08:
Wow!  The pictures look great, @Boguinhos !

Your measurements for voltage (presumably done when running all at all-on, maximum brightness) look great.  If you didn't have them all turned on, I suggest trying again with them all on and full-bright.  At 37 pixels per section with these numbers, I would be surprised to see instability caused by voltage.

> Power supply > + master fuse+ > + distributor > +section fuse+> section independant wire 1mm2 > strip+ > gnd wire back > gnd distributor > power supply -

While ASCII diagrams can't fully capture the layout, what I see above is something I would do ... **_for a single power supply_**.  Nice!

> I fused all the sections and the master connection to the power supply.  ... multiple power supply (3x10) ... have to have the gnd merged ... 
> The strip sections are not connected together on the strip itself. Only after fusing on the wiring.

Yes, merging of ground is the first step, and I also would not connect the strips directly.

Consider adding a **diode** at each place you connect a strip's ground to the common ground.  This can help ensure the strip isn't used as a ground path, except for what has come through its V+ side.  Adding a fuse at the same spot would be "belt-and-suspenders".  Of course, I would ensure the common ground wire is rated to handle the full current, use more fuses to validate my design, or have a electrician help design it.  After all, the 3x power supplies are generating up to 21.5A of power, and they are all sharing a single ground ... which means the ground might carry all 21.5A of power at places.  (simple scenario: ground connection to two of the power supplies gets disconnected).

Anyways, there's just an exponential number of failure modes when using more than one power supply.  It's not something I'd recommend normally.

> Thank you for all your help again

My pleasure to share what I (as an untrained random person) would do.  I think you're on the right path for single power supplies.  Multiple power supplies are an area outside my comfort zone at the moment ... so the above is just early thoughts.

Good luck, love the pictures ... it looks fabulous!

closed 2021-06-04 by stale[bot]